iamromulan/quectel-rgmii-toolkit #2: Does not work in rg520n.

EnzoLorio, 2024-01-22:
RG520NEUDBR01A06M8GA
/ # cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00280000 00040000 "sbl"
mtd1: 00100000 00040000 "mibib"
mtd2: 01600000 00040000 "efs2"
mtd3: 00280000 00040000 "rawdata"
mtd4: 00380000 00040000 "sys_rev"
mtd5: 00280000 00040000 "cust_info"
mtd6: 00200000 00040000 "tz"
mtd7: 00140000 00040000 "tz_devcfg"
mtd8: 00140000 00040000 "ddr"
mtd9: 00140000 00040000 "apdp"
mtd10: 00140000 00040000 "xbl_config"
mtd11: 00100000 00040000 "xbl_ramdump"
mtd12: 00140000 00040000 "multi_image"
mtd13: 00100000 00040000 "multi_image_qti"
mtd14: 00140000 00040000 "aop"
mtd15: 00140000 00040000 "qhee"
mtd16: 00140000 00040000 "abl"
mtd17: 00340000 00040000 "uefi"
mtd18: 00140000 00040000 "toolsfv"
mtd19: 00140000 00040000 "loader_sti"
mtd20: 00100000 00040000 "logfs"
mtd21: 00140000 00040000 "devinfo"
mtd22: 00140000 00040000 "sec"
mtd23: 00a80000 00040000 "multi_fota"
mtd24: 00140000 00040000 "misc"
mtd25: 00140000 00040000 "ipa_fw"
mtd26: 00140000 00040000 "usb_qti"
mtd27: 01700000 00040000 "boot"
mtd28: 01700000 00040000 "b_boot"
mtd29: 05b00000 00040000 "modem"
mtd30: 05b00000 00040000 "b_modem"
mtd31: 06500000 00040000 "system"
mtd32: 06500000 00040000 "b_system"
mtd33: 01f00000 00040000 "oemapp"
mtd34: 01f00000 00040000 "b_oemapp"
mtd35: 01f00000 00040000 "oemdata"
mtd36: 1ad80000 00040000 "usrdata"

`Archive:  simpleadminfull.zip
5c9e2ca77f3a67b1bbd2bed426d9fad8c7d6a295
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/scripts/build_modem_status
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/scripts/doAT.py
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/scripts/modemstatus_parse.sh
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/scripts/tojson.sh
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/systemd/simpleadmin_generate_status.service
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/systemd/simpleadmin_httpd.service
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/systemd/ttl-override.service
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/ttl/ttl-override
 extracting: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/ttl/ttlvalue
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/atcommander.html
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/cgi-bin/get_atcommand
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/cgi-bin/get_csq
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/cgi-bin/get_ttl_status
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/cgi-bin/set_ttl
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/css/admin.css
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/css/bulma.css
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/index.html
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/js/alpinejs.min.js
  inflating: quectel-rgmii-toolkit-simpleadminfull/simpleadmin/www/ttl.html
cp: cannot create regular file `/lib/systemd/system/simpleadmin_generate_status.service': Read-only file system
cp: cannot create regular file `/lib/systemd/system/simpleadmin_httpd.service': Read-only file system
cp: cannot create regular file `/lib/systemd/system/ttl-override.service': Read-only file system
ln: /lib/systemd/system/multi-user.target.wants/simpleadmin_httpd.service: Read-only file system
ln: /lib/systemd/system/multi-user.target.wants/simpleadmin_generate_status.service: Read-only file system
ln: /lib/systemd/system/multi-user.target.wants/ttl-override.service: Read-only file system
Failed to start simpleadmin_generate_status.service: Unit simpleadmin_generate_status.service not found.
Failed to start simpleadmin_httpd.service: Unit simpleadmin_httpd.service not found.
Failed to start ttl-override.service: Unit ttl-override.service not found.
Welcome to iamromulan's RGMII Toolkit script for Quectel RMxxx Series modems!`

iamromulan, 2024-01-23:
Looks like it's having trouble mounting / as read write. Can you attempt this manually? 

Also, this is meant for RMxxx m.2 modems. I have never tested this with an LGA form factor modem. First line beyond the title in the readme.md for the repo: "ORIGINALLY FOR THE QUECTEL RM520N-GL, However, People are saying this will work on all M.2 RMxxx modems"

iamromulan, 2024-03-16:
This Repo will focus on compatibility with m.2 from factor RM5xxx series modems for now.
feel free to fork and edit for your own use :)
**CLOSED**

closed 2024-03-16 by iamromulan