iamromulan/quectel-rgmii-toolkit #4: 5G NR-SA Bandwidth Values Not Parsed

danry25, 2024-02-07:
When connected to 5G-SA (5G Standalone), the Simple Admin bandwidth indicator is showing the increment between 0 and 14 then putting a Mhz tag on the end, but the value mappings (shown around page 97 in the [AT Commands Manual](https://www.waveshare.com/w/upload/8/8a/Quectel_RG520N%26RG52xF%26RG530F%26RM520N%26RM530N_Series_AT_Commands_Manual_V1.0.0_Preliminary_20220812.pdf) ) show that 12 means that you are connected to a 100Mhz channel.

Looks like [nr_bw is only called for NR-NSA](https://github.com/rbflurry/quectel-rgmii-simpleadmin/blob/main/scripts/modemstatus_parse.sh#L192) (5G Nonstandalone) causing this value to not be parsed.

[image]

iamromulan, 2024-02-08:
Seems so! Good catch!
Yep there are a few issues with it that were inherited from the original source (ROOTer)
The Unknown NR band in NSA every time is what was the most glaring for me. I will look at this a little closer and see what I can improve. In the meantime feel free to PR 😁

jpda, 2024-02-09:
@iamromulan I have been using your tools for a while, thank you for putting all of this together.

for what it's worth, I have a few scripts here https://github.com/jpda/local-bin/blob/main/openwrt (openwrt, openwrtps) that I used for parsing out qeng=servingcell and qcainfo. I was getting the AT response from the web interfaces of openwrt/rooter, but the parsing is pretty much the same regardless of the source.

dr-dolomite, 2024-03-16:
I will try to look on it since I already patched the bandwidth error for NSA NR bands.

dr-dolomite, 2024-03-16:
Update: missing function call for BW conversion. Fixed.
[image]

iamromulan, 2024-03-16:
**SOLVED**
Well done! Thank you for your contribution! Approved and merged!
https://github.com/iamromulan/quectel-rgmii-toolkit/pull/9

**CLOSED**

closed 2024-03-16 by iamromulan