iamromulan/quectel-rgmii-toolkit #7: IP Passthrough

dr-dolomite, 2024-03-13:
Good day! I just want to ask if it is possible to bridge the modem to router without the traffic being routed to 192.168.225.1? Im using IPPT and tracert shows that it still making my router behind NAT. I want to use GL inet's DDNS and it requires a direct public ip address and even though the dashboard shows my public ip address, it still doesnt work. Thanks!

[image]

iamromulan, 2024-03-14:
Hello there!
If you are getting assigned a public IP directly disable IPPT Nat. 

The way I look at it these modems have "3 stages of IP Passthrough"
First being the DMZ method
Then the MPDN rule
Finally true IPPT mode. 
Refer to this section
https://github.com/iamromulan/quectel-rgmii-configuration-notes?tab=readme-ov-file#disable-nat-to-vlan

Note that you will loose local network access to the gateway IP/simpleadmin/servers on the modem.

You can use Tailscale to get around that limitation.

dr-dolomite, 2024-03-16:
I see but Im not that confident messing with module configs right now especially since I cannot fully understand all of it despite reading the documentations. For now, I applied the DDNS patch for GL devices behind NAT. Thanks for this still. Cheers!

closed 2024-03-16 by dr-dolomite