iamromulan/quectel-rgmii-toolkit #14: wget error

GhostRider5666, 2024-03-20:
Resolving raw.githubusercontent.com... failed: Temporary failure in name resolution.
wget: unable to resolve host address 'raw.githubusercontent.com'

iamromulan, 2024-03-20:
Is the modem able to reach the internet? 
Is it a problem resolving hostnames, IP addresses, or both?
within the modems shell see if you can reach google. Try ping google.com and ping 8.8.8.8
Ctrl-C should stop the ping

GhostRider5666, 2024-03-20:
Yes, I can do all of that?

iamromulan, 2024-03-20:
If you indeed have Internet connectivity and can reach both 8.8.8.8 and google.com via ping that rules out any DNS issues.

Perhaps there really was some sort of temporary issue. 

Also I forgot to ask, is that error as result of trying to install something from the toolkit or trying to run the toolkit in general?

Within the modems shell try running this again, it may very well have been a temporary issue:
cd /tmp && wget -O RMxxx_rgmii_toolkit.sh https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/main/RMxxx_rgmii_toolkit.sh && chmod +x RMxxx_rgmii_toolkit.sh && ./RMxxx_rgmii_toolkit.sh && cd /

GhostRider5666, 2024-03-20:
I get the same thing after running adb shell?

GhostRider5666, 2024-03-20:
[image]

iamromulan, 2024-03-20:
Run the AT command AT+QMAPWAC?
What number does it respond with?

iamromulan, 2024-03-20:
> [image]
> 

Are you available on a private message platform? It's very odd that you can reach other sites but not GitHub. I'm willing to provide remote support to diagnose. Just download AnyDesk and have the modem connected by USB and give me the AnyDesk code. You'll want to private message me that though.

GhostRider5666, 2024-03-20:
Facebook?

iamromulan, 2024-03-20:
> Facebook?
> 

https://www.facebook.com/share/9rJNcaSKmsftfxcL/?mibextid=qi2Omg

iamromulan, 2024-03-20:
Issue solved by flashing firmware.
Tailscale messed with the resolv.conf file causing DNS issues.

Tailscale install process patched to prevent this in the future.

closed 2024-03-20 by iamromulan