iamromulan/quectel-rgmii-toolkit #25: Simplefirewall not working when in passthrough mode

snowzach, 2024-04-05:
The resources you have here are amazing.. Thank you so much! 

I've been playing with your toolkit on my RM521. I put my modem into passthrough mode and have confirmed that whatever ports are open on the modem are available on the internet via IPv6. I'm not sure what black magic is happening on the modem but it's bypassing the rules you put together by default. 

I am wondering if it would be better to only allow specific ip addresses (192.0.0.1 and 192.168.225.whatever)

snowzach, 2024-04-05:
Actually, IPv6 rules are different from IPv4 rules. It might be better to just listen on IPv4:8080 instead of all addresses (IPv6 also) otherwise we might need to install ip6tables to manage those.

iamromulan, 2024-04-05:
> Actually, IPv6 rules are different from IPv4 rules. It might be better to just listen on IPv4:8080 instead of all addresses (IPv6 also) otherwise we might need to install ip6tables to manage those. 

The intention is for the services accessible on the modem to be IPv4 only. httpd just listening on IPv4:8080 sounds good to me. 😁 We definitely want to avoid binding to specific IP addresses though as we are able to define a custom DHCP range and gateway address for the LAN. On top of that we have tailscale and potentially other future content that limiting to listening on a single IP would hinder.

The goal is eventually to tie everything together.
2 different webservers: one on 80 one on 443

80 is basic stats and a login button that redirects to 443

443 has SSL and authentication,
Once logged in full simpleadmin plus ttyd integrated into a new page (console)

However @dr-dolomite is hard at work with a much nicer front end and client-side AT response parsing. I wanted to give him some time before making any new changes there.

With the involvement of ttyd this also means we'll need to make entware a dependency as that's how we are replacing Quectel's custom login binary for something normal. I suppose entware isn't a must but it's guaranteed availability makes continued progress easier.

I'm honestly very happy to see that people are interested in contributing to the project. I have limited time to work on this so this definitely helps! Since we have started to move pretty quickly with new changes and additions I decided my area of focus for now should be on creating an automatic update system. However we can talk more about that and other potential movements in discussions. I think adding ip6tables to simplefirewall is a great idea.

snowzach, 2024-04-05:
I'm happy to help a little bit.. I don't have gobs of time either but I can tinker... Reverse proxies are my wheelhouse... 

Have you already started work on the webserver? Ideally simpleadmin and ttyd will listen on localhost (127.0.0.1) only and a reverse proxy (nginx) can listen to 80/443 and proxy to simple admin and ttyd under different paths (and handle the authentication bit) I'll work on that if you want...

dr-dolomite, 2024-04-05:
New simple admin front end might get finished in early May or end of April. Just a heads up.

snowzach, 2024-04-05:
That's cool...  I can put an nginx reverse proxy in front of it... Will the simepleadmin do auth or should nginx?

iamromulan, 2024-04-05:
nginx should do auth for now for sure,
With entware installed: opkg install nginx
looks like it installs nginx/1.25.3
Then /opt/etc/nginx is your configs

openssl is already built into the firmware as well.

If we are going to use nginx lets just abandon httpd completely and have nginx serve both the http and https content.

ttyd can move to port 8443 since 443 is going to be used by nginx instead

I will have to work on editing the toolkit script as well to make entware an installation dependency of simpleadmin

I think it's a good idea to add ipv6 to simplefirewall as well, its on my to-do list as well.

This is something I was going to work on later on down the road but my to-do list is already piled up for this project.
Thank you!!!!

snowzach, 2024-04-06:
Okay, I tinkered with it a bit tonight... So as it turns out you need a bunch of boilerplate crap to install nginx and make the cgi scripts work. I looked around at whatever else was available in opkg... There's an ancient version of lighttpd. I was able to setup a reverse proxy for simpleadmin and ttyd without too much effort running on https with a self-signed cert.. The only issue I had is lighttpd runs as user nobody (as it probably should) and the pty's that enable access to the AT console are owned by root. Will have to fix the permissions on those. After I did that it seemed to work pretty well. 

So yeah, I'll see if I can button this up and put a bow on it.. 

Basically:
* Configure simpleadmin and ttyd to listen on 127.0.0.1:8080 and some other port.. 
* Generate a self-signed cert
* Setup lighttpd to listen on 80 and 443 with some sort of auth and proxy to the other two. 

That will fix the firewall issue and clean up the access to the tools. I still don't love that it listens on all interfaces even if it's just IPv4... Be good if we could tighten that down to only listen to the lan port. There's still some funky bridging going on that could still allow the internet to access it if the IPv4 address was allowed in (those with static ip maybe, will have to check)

snowzach, 2024-04-06:
Do you happen to like ttyd vs just installing the openssh daemon?

iamromulan, 2024-04-07:
Apologies for the delayed response. 
Yeah ttyd works super well, and I'd definitely prefer to keep it. SSH is a great backup, however since tailscale covers that feature i didn't pursue it. 

You're right lighttpd looks like it's going to be a better choice. Honestly the lighttpd version didn't seem too old being from Oct 30th 2023 (1.4.73)

[image]

Do you think lighttpd-mod-access would let us set it up so only logged in users can see all pages instead of just the stats page? Originally I was thinking about separate instances where 80 is a www directory served and the 443 is a separate www directory served, however if that mod makes it so users are forced to login before accessing the content that would be ideal. Apologies I'm kind of new at this and I'm still learning, i appreciate the help.

I'll have to look into adding the opkg install commands to the sh toolkit. I'll also need to implement some more checks to make sure the current entware installation has the login, passwd, and .profile modifications made as well. Sounds like we will need useradd, usermod, userdel, and groupadd to be part of the stock login takeover as well. 

When I finish what I'm up to with the new simpleupdates installation, revision tracking will become easier. the toolkit sh will work together with it.

iamromulan, 2024-04-12:
@snowzach
The direct approach to address the original issue would be to add ip6tables to simplefirewall. I will look into this and close this issue once it is added.

iamromulan, 2024-04-13:
@snowzach That last PR fixed simpleadmin reading TTL but it cannot set TTL
-Toolkit can still set TTL though

closed 2024-04-16 by snowzach