iamromulan/quectel-rgmii-toolkit #28: HSTS

smashrvmco, 2024-04-06:
I get the following error when running the script.. doesnt seem to affect anything from not working. On my rm521f-gl this is mounted in a read-only format:

Will not apply HSTS. The HSTS database must be a regular and non-world-writable file.
ERROR: could not open HSTS store at '/home/root/.wget-hsts'. HSTS will be disabled.

Just wondering if I set something up wrong that could cause problems later down the road, or if this is expected behavior for now

dr-dolomite, 2024-04-06:
Those were completely fine afaik. If something went wrong, just use the script again to remove the installation, do a reboot and then install again.

iamromulan, 2024-04-07:
> Will not apply HSTS. The HSTS database must be a regular and non-world-writable file

https://superuser.com/a/1540028

> Failed to stop simplefirewall.service: Unit simplefirewall.service not loaded.
> Failed to stop ttl-override.service: Unit ttl-override.service not loaded.

These are ok

closed 2024-04-07 by iamromulan