iamromulan/quectel-rgmii-toolkit #34: Hi, I want to use the module in a "public" place, can I put a password on the web and telnet?

OvidioLee, 2024-04-17:
Hi, I want to use the module in a "public" place, can I put a password on the web and telnet?

Outside access I can modify the firewall to restrict access, but these two I really don't have a clue.

iamromulan, 2024-04-17:
That is actively being worked on.
https://github.com/iamromulan/quectel-rgmii-toolkit/discussions/30

You can try an early preview of this change by downloading the toolkit from the development branch but the TTL setting on simpleadmin is currently broken. It will view TTL but will not set it.
https://github.com/iamromulan/quectel-rgmii-toolkit/tree/development
Since Console (ttyd) is now integrated into simpleadmin as well you can easily set it through the toolkit script though.

Telnet? If you still have Telnet I would get rid of it. The micropython it used was just too much for the little single core armv7 (32-bit) processor. AT commands are a lot better now in simpleadmin and you can also send them through the console using the atcmd command.

iamromulan, 2024-04-17:
Also another solution is to turn on IP passthrough mode and disable IPPT NAT so the LAN/Gateway of the modem can no longer be reached locally. You'd still have access through tailscale if you had tailscale installed.