iamromulan/quectel-rgmii-toolkit #38: Hi can you please make offline version of simple admin

sevvian, 2024-04-25:
Hi Authors,

Again thanks for the team on the wonderfull work. I appreciate the teams effort.   I see the entire toolkit is made online like the modem to download it.

But i have a request (i tried to replace ,but i am not much of tech savvy) that anyway we can make the toolkit be offline version.

Say we can download the required repo to local laptop and push it via adb and then execute them locally in modem?  Or if you could let me know the places we have to replace variables to it local path, then we can do. 

Thanks,

iamromulan, 2024-04-29:
Are you having trouble with wget?

I hope to eventually create an offline deployment utility combined with the Auto installer found in my RM520 repo. It's my goal to make it into a full blown setup/deployment wizard. It'll do everything but flash base firmware, even then, if Qflash has command line parameters I might even be able to add that as well.

For now, the file structure of the GitHub is pretty much /usrdata

Eventually we will have this.

I'm going to go ahead and mark this as the solution and link it in the ideas discussion thread. 😁

closed 2024-04-29 by iamromulan

sevvian, 2024-05-02:
> Are you having trouble with wget?
>
> I hope to eventually create an offline deployment utility combined with the Auto installer found in my RM520 repo. It's my goal to make it into a full blown [...]

yeah, it would sound that i am noob :). when i ordered the mcuzone rj45 toolkit , i wrongly ordered the external power source with 3p power block auxilary power instead of Dc barrell input.   So now i find it hard to provide external power source through that power block or i dont know how. i do not want accidentally kill my rm521f .    

 now i have to power the rj45 kit with usb-c port and  ethernet out for internet .  through this mode working with AT is difficult as At command to be sent over ethernet, my only solution is to plug the usb-c  directly to laptop and run any AT commands without enabling internet. if i run internet directly  through usb-c , it would reboot in loop due to not enough power because usb-c is now running data and not pulling enough power from laptop. so running toolkit and let it download would be disastrous in my case, as it would break in the middle of script while downloading.

long story short, i could not run online .   It would be also handy to have offline version, pick choose only service we want. In my case i like simple admin  .   My scenario does not require simple firewall  and ttl exclusively, as i run my setup over vpn from my router.