iamromulan/quectel-rgmii-toolkit #43: 404 Error for Simple Network page

Alumni1506, 2024-05-08:
Hi,

Thank you for your excellent work. Installation of simple admin went smoothly.
The only issue I have is getting a "404 Not Found The requested URL was not found" whenever I try to access the "Simple Network" page.

[image]

[image]

dr-dolomite, 2024-05-09:
Marked as fixed. 
Reason: Incomplete install.

closed 2024-05-09 by iamromulan

Alumni1506, 2024-05-10:
Hi,

I reinstalled simpleadmin 5 times and I'm still getting the same error.
```
Installing Simple Admin...
What version of Simple Admin do you want to install? This will start a webserver on port 8080
1) Stable current version, (Main Branch)
2) Install Test Build (Development Branch)
3) Return to Main Menu
Select your choice:
1
1
Installing simpleadmin from the main stable branch
Removing installed AT Socat Bridge services...
Installing AT Socat Bridge services...
Connecting to raw.githubusercontent.com (185.199.111.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-armel-static'
socat-armel-static   100% |********************************|  864k  0:00:00 ETA
'socat-armel-static' saved
Connecting to raw.githubusercontent.com (185.199.111.133:443)
wget: note: TLS certificate validation not implemented
saving to 'killsmd7bridge'
killsmd7bridge       100% |********************************|   141  0:00:00 ETA
'killsmd7bridge' saved
Connecting to raw.githubusercontent.com (185.199.109.133:443)
wget: note: TLS certificate validation not implemented
saving to 'atcmd'
atcmd                100% |********************************|  1775  0:00:00 ETA
'atcmd' saved
Connecting to raw.githubusercontent.com (185.199.109.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-smd11.service'
socat-smd11.service  100% |********************************|   364  0:00:00 ETA
'socat-smd11.service' saved
Connecting to raw.githubusercontent.com (185.199.109.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-smd11-from-ttyIN.service'
socat-smd11-from-tty 100% |********************************|   301  0:00:00 ETA
'socat-smd11-from-ttyIN.service' saved
Connecting to raw.githubusercontent.com (185.199.110.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-smd11-to-ttyIN.service'
socat-smd11-to-ttyIN 100% |********************************|   301  0:00:00 ETA
'socat-smd11-to-ttyIN.service' saved
Connecting to raw.githubusercontent.com (185.199.111.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-killsmd7bridge.service'
socat-killsmd7bridge 100% |********************************|   164  0:00:00 ETA
'socat-killsmd7bridge.service' saved
Connecting to raw.githubusercontent.com (185.199.109.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-smd7-from-ttyIN2.service'
socat-smd7-from-ttyI 100% |********************************|   299  0:00:00 ETA
'socat-smd7-from-ttyIN2.service' saved
Connecting to raw.githubusercontent.com (185.199.108.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-smd7-to-ttyIN2.service'
socat-smd7-to-ttyIN2 100% |********************************|   299  0:00:00 ETA
'socat-smd7-to-ttyIN2.service' saved
Connecting to raw.githubusercontent.com (185.199.110.133:443)
wget: note: TLS certificate validation not implemented
saving to 'socat-smd7.service'
socat-smd7.service   100% |********************************|   365  0:00:00 ETA
'socat-smd7.service' saved
Adding AT Socat Bridge systemd service units...
AT Socat Bridge service online: smd11 to ttyOUT
AT Socat Bridge service online: smd7 to ttyOUT2
AT Socat Bridge services Installed!
Failed to stop simplefirewall.service: Unit simplefirewall.service not loaded.
[... 93 lines cut ...]
wget: note: TLS certificate validation not implemented
saving to 'bulma.css'
bulma.css            100% |********************************|  202k  0:00:00 ETA
'bulma.css' saved
Connecting to raw.githubusercontent.com (185.199.111.133:443)
wget: note: TLS certificate validation not implemented
saving to 'get_atcommand'
get_atcommand        100% |********************************|  1059  0:00:00 ETA
'get_atcommand' saved
Connecting to raw.githubusercontent.com (185.199.108.133:443)
wget: note: TLS certificate validation not implemented
saving to 'get_csq'
get_csq              100% |********************************|   210  0:00:00 ETA
'get_csq' saved
Connecting to raw.githubusercontent.com (185.199.110.133:443)
wget: note: TLS certificate validation not implemented
saving to 'get_ttl_status'
get_ttl_status       100% |********************************|   318  0:00:00 ETA
'get_ttl_status' saved
Connecting to raw.githubusercontent.com (185.199.109.133:443)
wget: note: TLS certificate validation not implemented
saving to 'set_ttl'
set_ttl              100% |********************************|  1349  0:00:00 ETA
'set_ttl' saved
```

Alumni1506, 2024-05-10:
Hi thanks for getting back.
Any attempt to install the development branch is refused it say it is under development

dr-dolomite, 2024-05-10:
> Hi thanks for getting back. Any attempt to install the development branch is refused it say it is under development

Go to development branch and use the link from it to download the installer.

Alumni1506, 2024-05-10:
**Update**
Installing simpleadmin via the development branch works. Thanks

iamromulan, 2024-05-10:
@Alumni1506 
Apologies, This is now fixed on the main branch https://github.com/iamromulan/quectel-rgmii-toolkit/commit/3962c5cf5a433a1bf66048d3a4343a0eb53d7b09

@dr-dolomite 
 support for Simple Admin v1 is about to end but not 100% until its pushed to main 😄

dr-dolomite, 2024-05-11:
> @Alumni1506 Apologies, This is now fixed on the main branch [3962c5c](https://github.com/iamromulan/quectel-rgmii-toolkit/commit/3962c5cf5a433a1bf66048d3a4343a0eb53d7b09)
> 
> @dr-dolomite support for Simple Admin v1 is about to end but not 100% until its pushed to main 😄

My bad 😆

Alumni1506, 2024-05-12:
No problem. When i get the chance, I'll test the main branch installation.