iamromulan/quectel-rgmii-toolkit #54: Feat: Automatic Band Selection based on latency

danry25, 2024-05-21:
Occasionally I run into these modems doing fun things like aggregating congested LTE Band 12 with LTE Band 2 and 5G N41 causing 250ms to 500ms latency spikes (whereas most traffic only experiences 20ms to 30ms of latency).

It would be handy to have a watchdog that runs a ping to 1.1.1.1, and when this latency exceeds a certain boundary (say 100ms) remove the band that is problematic from the list of allowed LTE, 5G-NSA and 5G-SA bands.

dr-dolomite, 2024-05-22:
Thats quite hard to implement since different Module versions (even variants) have different support for CA band combinations. But Ill try my best to create a similar approach on the Simple Watchcat.

iamromulan, 2024-05-23:
The next question is how would we determine what the "problematic" band is? Order of operations for removing bands? Re-ping, repeat?

closed 2024-05-23 by iamromulan