iamromulan/quectel-rgmii-toolkit #60: AT Commands not working

danry25, 2024-05-29:
Trying to send any AT command hangs at this page:
[image]
get_atcommand never responds:
[image]

**Environment**
Running current stable SimpleAdmin with just Tailscale installed, Tested in Firefox & Chrome on Debian 12, validated with a second modem.

dr-dolomite, 2024-06-03:
I suggest to use to development branch. It is much stable (simpleadmin) and received a lot of fixes.

MickMorley, 2024-06-19:
Hi @dr-dolomite, I found your Simple Admin 2.0 yesterday. It looks great! When will the changes in the development be pushed to the main branch?

danry25, 2024-08-29:
I tracked this down to /dev/ttyOUT2 not responding on some RM520N-GL's even after reflashing the firmware with QFirehose, it appears the wait time just endlessly ratchets up as minicom is attempted again and again.

If we could switch to /dev/ttyOUT if /dev/ttyOUT2 does not respond (eg: [when line 28 is run](https://github.com/iamromulan/quectel-rgmii-toolkit/blob/3962c5cf5a433a1bf66048d3a4343a0eb53d7b09/simpleadmin/www/cgi-bin/get_atcommand#L28)) this would mitigate the underlying firmware bug.

Here is where /dev/ttyOUT2 is called:

https://github.com/iamromulan/quectel-rgmii-toolkit/blob/3962c5cf5a433a1bf66048d3a4343a0eb53d7b09/simpleadmin/www/cgi-bin/get_atcommand#L22

iamromulan, 2024-10-30:
Main branch is archive and directory only now.
These issues don't exist on the development branch.
See the updated readme in main

closed 2024-10-30 by iamromulan

danry25, 2025-02-07:
This bug persists in the current SDXLEMUR branch, to fix it edit DEVICE=/dev/ttyOUT2 in /usrdata/socat-at-bridge/atcmd to drop the two.

Symptoms include hanging after running an AT command using the atcmd prompt in adb shell, and no values being displayed in SimpleAdmin 2.0