iamromulan/quectel-rgmii-toolkit #61: Restart fails to restart webserver

tarunVreddy, 2024-06-03:
On the development branch, any one else getting the web interface not running after restarting the sled?

dr-dolomite, 2024-06-04:
This was fixed with the current version I believe. Please update.

tarunVreddy, 2024-06-04:
At the time of the comment, I had done a fresh install. But I'm reinstalling firmware (RM520NGLAAR03A02M4GA_2023_11_06) and trying and install again. Not sure if that fixed it or it just takes a while for all the stats to come in after it boots up. Seems to work. Thank you!

iamromulan, 2024-06-09:
This issue still exists.
I'm about to go on vacation for 2 weeks, maybe between sight seeing I can look into this a little more.

Yes. The webserver sometimes does not start at boot for some reason. Roll of the dice. I suspect it's attempting to start it sometimes before opt.mount becomes available. Systemd on these modems have really strange behavior where some functions simply don't work. Systemctl enable for example does not work. Have to manually symbolically link.

iamromulan, 2024-10-30:
Will take a look at implementing the fix. If the system allows it we may try to do an overlayfs and pivot-root to the overlay. 
I'd have to get an opkg built that installs to / instead of /opt in that scenario but it'll be better

closed 2024-10-30 by iamromulan