iamromulan/quectel-rgmii-toolkit #88: How to access Simple admin after install?

drewmeel, 2024-10-28:
Hello I have a RM520N-GL and I have installed the main branch with the simple admin and the Entware/OPKG packages. yet when I go over to my browser and navigate to the http://192.168.225.1:8080/ ip address, simple admin does not load.

Just to let you know I am using just a usb enclosure not the one with a Ethernet port. before I installed Simple admin I could connect to my carrier with the same usb enclosure. but now I cant access the internet and also the simple admin. 

Thanks

iamromulan, 2024-10-30:
I have edited the main readme
https://github.com/iamromulan/quectel-rgmii-toolkit

Flash your RM520N-GL to the latest firmware the proceed with the directions from main

What are you using the USB enclosure with? This repo and directions are intended for one with a Ethernet port but you can use some of it.

First off sounds like your usbnet mode is going to need to chnage back to what it needs to be for the router you are using
See: https://github.com/iamromulan/RM520N-GL?tab=readme-ov-file#usb-modes

Since you will not need PCIe RC mode probably best to stay in EP mode ``AT+QCFG="pcie/mode",0``

Depending on the usbnet modes you are able to use will determine if you'll need to exclusively need to use tailscale instead. ECM and RNDIS will present the LAN of the modem to the USB host device while QMI/NDIS and MBIM will put the USB host device in charge and pass the external IP directly to it. In QMI/NDIS and MBIM modes you will need to use tailscale to access the deployed services on the modem.

McMacaron, 2024-10-31:
The same here. I can not connect to Havens technologie modem with http://192.168.225.1:8080/.
I can not install Tailscale:
Configure Tailscale
1) Enable Tailscale Web UI at http://192.168.225.1:8088 (Gateway on port 8088)
2) Disable Tailscale Web UI
3) Connect to Tailnet
4) Connect to Tailnet with SSH ON
5) Reconnect to Tailnet with SSH OFF
6) Disconnect from Tailnet (reconnects at reboot)
7) Logout from tailscale account
8) Return to Tailscale Menu
Enter your choice: 1
Connecting to raw.githubusercontent.com (185.199.109.133:443)
wget: note: TLS certificate validation not implemented
wget: server returned error: HTTP/1.1 404 Not Found
wget: write error: Broken pipe
Connecting to raw.githubusercontent.com (185.199.111.133:443)
wget: note: TLS certificate validation not implemented
wget: server returned error: HTTP/1.1 404 Not Found
wget: write error: Broken pipe
Tailscale Web UI Enabled
Starting Web UI...
Failed to start tailscale-webui.service: Unit tailscale-webui.service not found.
Web UI started!
Configure Tailscale

and finally I can not use zerotier as I do not have permission to write for bin folder 
/ # opkg install zerotier
/bin
ln -sf /opt/bin/zerotier-cli /bin
ln -sf /opt/bin/zerotier-idtool /binPackage zerotier (1.14.0-1) installed in root is up to date.
/ # ln -sf /opt/bin/zerotier-one /bin
ln: /bin/zerotier-one: Read-only file system
/ # ln -sf /opt/bin/zerotier-cli /bin
ln: /bin/zerotier-cli: Read-only file system
/ # ln -sf /opt/bin/zerotier-idtool /bin
ln: /bin/zerotier-idtool: Read-only file system
/ # ln -sf /opt/bin/zerotier-one /bin
ln: /bin/zerotier-one: Read-only file system

I am bit stacked here. Everything went so nice in 30min I had internet and now I have spend another 5 hours going nowhere.

iamromulan, 2024-10-31:
I haven't tested Zerotier, I plan on making it an option from the menu soon. 

The tailscale webui enabler is now fixed. I forgot there were a few dependent files still linked to the main branch.

Also no 8080
Simpleadmin 2.0 is served through lighttpd on 80 which redirects 443 

- The config notes guide needs updated

McMacaron, 2024-10-31:
The same which you recommended. Just ordered from USA.
https://store.thewirelesshaven.com/products/rj45-usb3-poe-m2-modem-adapter-v8?variant=44407334371518

McMacaron, 2024-10-31:
It looks that servis started but I am not able to connect to tailscale on http://192.168.225.1:8088/

You are right simple admin is available with https://192.168.225.1. No ports needed

McMacaron, 2024-10-31:
Still it is amazing work. Thank you!!!!!

iamromulan, 2024-11-01:
> Still it is amazing work. Thank you!!!!!

Credit for simpleadmin 2.0 goes to @dr-dolomite 

Just checked the tailscale web UI looks like it's working 👍

[image]

McMacaron, 2024-11-01:
I am struggling with command line syntax. Where I can find list of commands or some documentation to system?

iamromulan, 2024-11-01:
> I am struggling with command line syntax. Where I can find list of commands or some documentation to system?

Probably on a desk somewhere at Quectel or Qualcomm HQ 🤣
Probably reserved for those high paying clients who buy enough units to be eligible for the SDK (QuecOpen)

I poked at the system and collected enough info about how it works until I had enough to work with. 

It's an OpenWRT system with an Android kernel if that is any help.

What commands are you having trouble with?

drewmeel, 2024-11-05:
@iamromulan thank you for the update. I have gone ahead and ordered a NGMFF enclosure with RJ45 port and I'll report back to what my results are. 

One more question for you is: does the simple admin for RM520N-GL have IP Passthrough? Or is it only for the newer SDX75 modems? As I've seen a preview of IP passthrough on quecmanager gui.

Thanks again

drewmeel, 2024-11-10:
Hello, an UPDATE:

So I have installed the RM520 in a M.2 to Ethernet [enclosure](https://www.aliexpress.us/item/3256807785730018.html) and I get the connection after installing the simple admin and also the entware package. However whenever I try to go to 192.168.225.1:8080 on all of my browsers it wont direct me to the simple admin gui. 

Also the enclosure has Realtek RTL8125BG Controller . I also ran the first time setup AT commands. 

```
AT+QCFG="data_interface",0,0
AT+QETH="eth_driver","r8125",1
AT+QCFG="pcie/mode",1
AT+QCFG="usbnet",1  #[if you need to be in a different USB mode for whatever reason remember to send AT+QMAPWAC=1]
AT+QMAPWAC=1  #[optional if in ECM mode, as this will happen on its own anyways.]
AT+CGDCONT=1,"IPV4V6","apn-here-inside-of-quotes"
Any other *magic* commands you want to run for proper cellular connection
AT+QPRTPARA=1 [optional]
AT+CFUN=1,1
```

So anything im doing wrong?

dr-dolomite, 2024-11-10:
> Hello, an UPDATE:
>
> So I have installed the RM520 in a M.2 to Ethernet [enclosure](https://www.aliexpress.us/item/3256807785730018.html) and I get the connection after installing the simple admin and [...]

No need to specify the 8080 port.

drewmeel, 2024-11-10:
> No need to specify the 8080 port.

Even that won't reach the gui

drewmeel, 2024-11-27:
nothing will work guys.

iamromulan, 2024-11-28:
It's just https://192.168.225.1/

Updated the guide a little tonight but it's in need of refinement and updates.

iamromulan, 2024-12-04:
I'll make a video for the 520 some point soon 👍

drewmeel, 2024-12-04:
> I'll make a video for the 520 some point soon 👍

thanks, take your time. I appreciate it greatly

iamromulan, 2025-01-13:
Closed for now. I still plan to make a video it just may be awhile. I'm redesigning SDXLEMUR slowly but SDXPINN is the current priority.

closed 2025-01-13 by iamromulan