iamromulan/quectel-rgmii-toolkit #89: Unable to login QuecManager

WTa-hash, 2024-11-01:
On 2024-NOV-1, I went through setting up (in order):

1. sdxpinn-mount-fix
2. Install basic packages/enable Luci
3. Set Root password
4. Set TTL
5. Install QuecManager 0.0.3

I can login to Luci using the password configured in step 3. **However, I cannot login to QuecManager (http://192.168.224.1/index.html) using the same credentials.**

[image]

How can I access QuecManager?

WTa-hash, 2024-11-06:
Ok - the issue is because my password contains special characters like !@$

dr-dolomite, 2024-11-06:
> Ok - the issue is because my password contains special characters like !@$

Thats good to know. Encode URL might be bugging out when having special characters as your input pass. Will consider this to improve the login script.

iamromulan, 2025-01-13:
@dr-dolomite Has this been fixed? I thought I noticed % working as a password.

dr-dolomite, 2025-01-19:
I honestly forgot about it but I will rewrite the script to support limited special characters for a password.

iamromulan, 2025-01-20:
This has been fixed in the current beta 1.1.1-1
1.1.1 is the release candidate and will be merged into the main package after more testing and confirmation.
You may reopen this issue if the problem still occurs.

closed 2025-01-20 by iamromulan