iamromulan/quectel-rgmii-toolkit #90: Set MTU for modem

WTa-hash, 2024-11-01:
I am currently using the RM551E with m.2 to RJ45 adapter (RC PCIe). How do I configure the MTU for the modem to 1428?

iamromulan, 2024-11-10:
I will look into this. I've never needed to edit MTU before. What's the benefit?

I believe each interface has an mtu which you can see by running the ifconfig command.

WTa-hash, 2024-11-10:
T-Mobile and Verizon MTU are around 1420. I believe by default each interface are set to 1500. The benefit is to reduce fragmentation.

iamromulan, 2024-11-19:
There are typically 2 interfaces related to the cellular connection seen active in ifconfig (including each MTU).
An rmnet_data interface and rmnet_ipa0
Here is how it works:

- Data Path Interface: rmnet_ipa is typically the direct connection to the IP accelerator, managing the data processed by the IPA hardware.
- High-Speed Data Transfer: It often supports large MTUs (e.g., 9216 bytes) to optimize throughput, especially for scenarios requiring high-speed data handling, such as 5G.
- Secondary to rmnet_data Interfaces: While rmnet_dataX interfaces provide data paths for individual data sessions (e.g., PDP contexts), rmnet_ipa often serves as the aggregated backend for all rmnet_dataX interfaces.

So in theory we only need to modify the MTU of the rmnet_data interface?
Take a look at my my rmnet interfaces
```
root@RM551EGL:~# ip link show | grep -A1 rmnet
14: rmnet_ipa0: <UP,LOWER_UP> mtu 9216 qdisc mq state UNKNOWN mode DEFAULT group default qlen 1000
    link/[519]
--
18: rmnet_data0@rmnet_ipa0: <> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/[519]
19: rmnet_data1@rmnet_ipa0: <UP,LOWER_UP> mtu 1430 qdisc mq state UNKNOWN mode DEFAULT group default qlen 1000
    link/[519]
20: rmnet_data2@rmnet_ipa0: <> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/[519]
21: rmnet_data3@rmnet_ipa0: <> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/[519]
22: rmnet_data4@rmnet_ipa0: <> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/[519]
23: rmnet_data5@rmnet_ipa0: <> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/[519]
24: rmnet_data15@rmnet_ipa0: <> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/[519]
25: rmnet_data16@rmnet_ipa0: <> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/[519]
```

Here we can see rmnet_data1 is UP with an MTU of 1430
If I wanted to edit that I would probably send a shell command like ```ip link set rmnet_data1 mtu 1420```

closed 2024-11-20 by WTa-hash

JackCA, 2024-11-20:
How would I automate this on startup?

iamromulan, 2024-12-07:
MTU modification is now available from the Toolkit/deployment script. It does persist at boot and gets placed in the custom firewall rules right along with TTL 👍
[image]

dr-dolomite, 2024-12-07:
This will also be available on QuecManager itself soon.

JackCA, 2024-12-09:
Awesome!