iamromulan/quectel-rgmii-toolkit #96: I CANT ACCESS WEB INTERFACE FOR SECURITY ONION AFTER CONFIGURATION

doughtyben89, 2024-12-11:
Hi everyone, I am new to Security Onion please I need help, I have successfully install and configure evac mode of security onion on Oracle VirtualBox Manager, and I used DHCP as the network connection option. when I launched the machine and log in with administrator username and password, it gave me the IP address to login to the web interface but when I use the IP address to access the web interface it does connect. I check the status of my security onion everything runs fine. please see image if there is anything you can help me with. Thank you for your help

[image]
[image]

cris-renee, 2025-01-06:
Hi I'm having the same issue and was wondering if you figured out a solution?

iamromulan, 2025-01-12:
I'm honestly not too familiar with this. Your trying to get to the web interface of the modem through this?

iamromulan, 2025-01-13:
Closed. Not related to project.

closed 2025-01-13 by iamromulan