iamromulan/quectel-rgmii-toolkit #109: Few problems and few suggestions:

d5aqoep, 2025-01-19:
1. Right now, QuecManager UI auto chooses absolutely the wrongest speedtest server in a different city. Is it possible to set own speedtest server by using server ID from speedtest.net website? The ID which we can see if we hover on ISP name while choosing the speedtest server manually on the website?

2. The tailscaled 1.78.1 package is totally busted in your repo. I had to remove iamromulan repo and install an old Tailscale 1.32 version which is there in default OpenWRT repo. Then use the below terminal script to update it to latest version.

wget -O update-tailscale.sh https://raw.githubusercontent.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh \
&& sh update-tailscale.sh

tailscale set --auto-update

3. Signal Metrics are wrong and they don't match with the real metrics shown for Current Active Bands.

But things are progressing nicely and I liked the new QuecManager UI. Hoping for new firmware with better 5G NR speeds.

dr-dolomite, 2025-01-19:
1. A custom speedtest page with the ability to select a custom sever is under planning and I'll try my best to implement it. However, for now, other features are in higher priority so it might take some time.

2. Let's wait for @iamromulan's reponse.

3. Signal Metrics were made to simply show an absolute average based on the signal metrics output from each of the port through AT+QRSRP, AT+QRSRQ, and AT+QSINR. There's no other plans to change this approach. For an accurate signal reading, another feature is under planning for it.

iamromulan, 2025-01-19:
Totally busted? I'm not too sure what you mean. I did recently merge some files from luci-app-tailscale into the tailscaled package and now the config defaults to not having an instance run. I'll go ahead and edit the default config to have it be enabled. For the best experience installing tailscale from the repo install the luci-app-tailscale package.

It requires a special version of the init.d and config file so you'll want to uninstall the tailscale package you have currently installed.

d5aqoep, 2025-01-20:
[img]https://i.imgur.com/RnS8rLq.png[/img]
```
Errors

uci: Entry not found
Collected errors:
 * file_sha256sum_alloc: Failed to open file /# Conff file needs replaced for now: No such file or directory.
 * file_sha256sum_alloc: Failed to open file /# /etc/config/tailscale: No such file or directory.
```

iamromulan, 2025-01-20:
- Dedicated Speedtest page is on the list
- Tailscale and Tailscaled should be fine now

closed 2025-01-20 by iamromulan