iamromulan/quectel-rgmii-toolkit #111: Failed packages

mailz23, 2025-01-20:
* opkg_conf_parse_file: Duplicate src declaration (iamromulan-SDXPINN-repo https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/opkg-feed). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (iamromulan-SDXPINN-repo https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/opkg-feed). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (iamromulan-SDXPINN-repo https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/opkg-feed). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (iamromulan-SDXPINN-repo https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/opkg-feed). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (iamromulan-SDXPINN-repo https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/opkg-feed). Skipping.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/targets/sdx75/generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiagm/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiargs/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiaudio/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiaudioprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtibsp/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtibspprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtibt/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtibtprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qticore/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qticoreinternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qticoreprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtictainternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtidata/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtidatainternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtidataprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiipq/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiipqezmesh/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiipqprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtilocation/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtilocationinternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtilocationprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtimsdcprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtinternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtipal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiperfprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtippatprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiquectel/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiquectelprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtirilprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtisecurity/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtisecurityinternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtisecurityprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtisensorsinternal/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtisensorsprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtissdk/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtissmgr/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtissmgrprop/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiwlan/Packages.gz, wget returned 8.
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/qtiwlanprop/Packages.gz, wget returned 8.

iamromulan, 2025-01-20:
That's pretty strange,
Its the toolkits job to list the repo one time. Have you ran the toolkit multiple times?

You can edit the custom feeds by going into Luci Software --> Configure opkg

mailz23, 2025-01-20:
I forgot to capture it the first time so I ran it again. Those same packages failed.

iamromulan, 2025-01-20:
Those are package sources.
I would just delete the extra entrys in the customfeeds and you should be all good

mailz23, 2025-01-21:
So those packages from openwrt release packages aren't needed? Just want to confirm before I delete it.

iamromulan, 2025-01-21:
The firmware before the December firmware had some just like that so I added shell scripting to hide those non-working ones. This one i guess has more. 

So when an OpenWRT system is built a sources list is automaticly generated and placed in the system. You can see it when you click configure opkg. If you notice when you run opkg update it reads that source file to update its own index of packages available to it to install. A lot of those don't exist. Proprietary package repos from Qualcomm or Quectel don't usually get added to the public openwrrt repo server. The pre-installed packages though are associated with each one of the repos on the list in some way. 

I will make a new package called sdxpinn-patch

I'll include fixes like this in there from now on. 

My dilemma is: I could include it in my mount-fix package like I did before, but that's really only meant to be installed one time after a fresh flash. So the idea is I can have a package with a general system patch that will make sure certain things are set correctly for everyone at once.

I'll close this issue when I'm done.

iamromulan, 2025-01-22:
sdxpinn-patch is now available. Go ahead and install it.

closed 2025-01-22 by iamromulan