iamromulan/quectel-rgmii-toolkit #113: [Feature request]: Display frequency in addition to bands

hermsi94, 2025-01-24:
Is it possible to display the frequency in quecmanager in addition to the band used, as described here:
https://en.m.wikipedia.org/wiki/LTE_frequency_bands
https://en.m.wikipedia.org/wiki/5G_NR_frequency_bands

? This would make it even easier to assign the frequencies used and you wouldn't always have to look up which frequencies are currently being used in Wikipedia.

[image]

dr-dolomite, 2025-01-24:
Sorry but I dont think that this feature will be viable or else, reliable in any way. Its not a 1:1 conversion and the modem doesnt provide native and direct information about it except for E/ARFCN. For now, you can use E/ARFCN calculators and it could produce estimations of its frequency.

hermsi94, 2025-01-24:
> Sorry but I dont think that this feature will be viable or else, reliable in any way. Its not a 1:1 conversion and the modem doesnt provide native and direct information about it except for E/ARFCN. For now, you can use E/ARFCN calculators and it could produce estimations of its frequency.

Okay. then I'll use the calculator. I didn't know until now that I could get more precise information via the E/ARFCN.

Perhaps a calculator can be integrated offline in quecmanager at some point.

Regards

dr-dolomite, 2025-03-13:
NR/E/ARFCN Calculator was added to the latest beta 2.0.1. Closing this now.

closed 2025-03-13 by dr-dolomite

hermsi94, 2025-03-14:
Very nice. The cell search with the values, displayed frequencies etc. is really great. Thank you. 👌

Many thanks for your efforts

dr-dolomite, 2025-03-14:
Really appreciate that feed back 😁