iamromulan/quectel-rgmii-toolkit #115: RM551 issue with installing packages in openwrt

borfoo3, 2025-01-27:
Hello

I have openwrt up and rumning on the 5g2phy with rm551 running latest firmware Dec 2024. 
Tailscale is also installed and shows as working.

I am having a problem with installing the speedtest and the other packages from your video, by getting the following error code 255. Any advise?

[image]

iamromulan, 2025-01-29:
Something is wrong if it can't find libc

You should most definitely flash fresh and try again.

https://mega.nz/file/7IVVGDpa#W0nGKnVz0WW9tEQI3W3ndm-h39X40imQjPnSfh7FJCI

Run the toolkit over ADB after that like normal 
Install the Mount fix first
Set TTL and MTU if needed
Install my repo/basic packages
From there should be all good 👍

borfoo3, 2025-02-04:
Ok so it finally worked as I re-did the steps.

2 important lessons:
1- disable all antivirus in background
2- internet connection was required for firmware re-flash in QFlash for whatever reason

Now all good and working!