iamromulan/quectel-rgmii-toolkit #121: Console/ttyd not work

XiaoHaoJun, 2025-02-10:
I have tried reinstalling but still can't use the console/ttyd. (SDXLEMUR Branch)

[image]

[image]

svandive, 2025-02-11:
For me, the console simply never loads, I see a black screen with a white "Cursor" but nothing else.  This had been working fine for me, but I've also installed and reinstalled several times as I've had to wipe my modems during testing.

iamromulan, 2025-05-15:
Is this still an issue with the SDXLEMUR branch? It seems to be working ok for me.

clndwhr, 2025-05-15:
Works for me on latest firmware