iamromulan/quectel-rgmii-toolkit #123: 404: Not Found   https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/RM55x_rcPCIe_toolkit.sh

Qiea, 2025-02-19:
This sh script cannot be downloaded

https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/RM55x_rcPCIe_toolkit.sh

iamromulan, 2025-02-20:
https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/rcPCIe_SDXPINN_toolkit.sh

iamromulan, 2025-02-20:
> This sh script cannot be downloaded
> 
> https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/RM55x_rcPCIe_toolkit.sh
> 

Where did that link come from? 🤣

Qiea, 2025-02-20:
> > This sh script cannot be downloaded
> > https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/RM55x_rcPCIe_toolkit.sh
> 
> Where did that link come from? 🤣

https://github.com/iamromulan/quectel-rgmii-toolkit/blob/SDXLEMUR/RMxxx_rgmii_toolkit.sh
Line 753 of this file

This was discovered while installing scripts for the RM500U module

By the way, I originally intended to close this issue, but I couldn't find the delete button🤣

closed 2025-02-20 by Qiea

SpeedxThunder001, 2025-02-20:
same error here

can you update it please?

@iamromulan

dr-dolomite, 2025-02-20:
> same error here
> 
> can you update it please?
> 
> [@iamromulan](https://github.com/iamromulan)

Its not something that we could easily fix since the compatability of SimpeAdmin / QuecManager depends on the architecture of the modem. As of now, Unisoc based modems are not supported.

SpeedxThunder001, 2025-02-20:
oh so i have rm551e and simple admin is not ported for that architecture?

dr-dolomite, 2025-02-20:
> oh so i have rm551e and simple admin is not ported for that architecture?

My bad. I assumed that you have the same modem as the OP. Install scripts works fine on my end for RM551E. Can you confirm if there is an internet connection through ping?

SpeedxThunder001, 2025-02-20:
yes everything works

i was thinking to install simple admin instead of quecmanager but i think both are same, right?

dr-dolomite, 2025-02-20:
> yes everything works
> 
> i was thinking to install simple admin instead of quecmanager but i think both are same, right?

No. Simpleadmin has been discontinued in favor of QuecManager and only QuecManager will work with RM551E. We will port QuecManager to other models once it reaches a stable version.

SpeedxThunder001, 2025-02-20:
oh ok

also i installed the beta version of quecmanager but i am getting this error

https://ibb.co/67sXsRBH

SpeedxThunder001, 2025-02-20:
@iamromulan @dr-dolomite did you try quecmanager 1.1.1 in newest firmware A2?

it seems there are some issue with it

SpeedxThunder001, 2025-02-20:
[image]