iamromulan/quectel-rgmii-toolkit #124: RM500U Does not support systemctl

Qiea, 2025-02-19:
root@udx710-module:/ # opkg update && opkg install systemctl
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Unknown package 'systemctl'.
Collected errors:
 * opkg_install_cmd: Cannot install package systemctl.

root@udx710-module:/ # cat /etc/os-release
ID=unisoc-initgc
NAME="unisoc-initgc-distro"
VERSION="udx710-module+unisoc-initgc-1.0+W24.10.4:14.56.00+user+native (sumo)"
VERSION_ID=udx710-module-unisoc-initgc-1.0-w24.10.4:14.56.00-user-native
PRETTY_NAME="unisoc-initgc-distro udx710-module+unisoc-initgc-1.0+W24.10.4:14.56.00+user+native (sumo)"
DISTRO_CODENAME="sumo"
root@udx710-module:/ # uname -a
Linux udx710-module 4.14.98 #1 SMP PREEMPT Thu Mar 7 07:12:10 UTC 2024 aarch64 GNU/Linux

dr-dolomite, 2025-02-19:
I dont think that it supports Unisoc release modules?

iamromulan, 2025-02-19:
RM or RG?

Doesn't appear to be SDXLEMUR, SDXPRAIRIE, or SDXPINN which is what this repo supports.

Confirmed:
(*Assumed not tested)

SDXPRAIRIE:
RM500Q*, RM502
SDXLEMUR:
RM520/521/530*
SDXPINN:
RM550*, RM551

However, your goal seems to be to understand what handles services and/or system init. Could be systemd (`systemctl `), could be systemV (like on OpenWRT; init.d [` service ` command] ), could even be BusyBox init. Just depends on how the system was built.

The kernel log right after it's booted can be helpful 
`dmesg`

Also understanding the mount points and storage allotment can also be helpful 

`df -h && mount`

Qiea, 2025-02-20:
It should be a model issue, this module is not supported

closed 2025-02-20 by Qiea