iamromulan/quectel-rgmii-toolkit #125: Using RM551E-GL in M.2 to RJ45 enclosure issues

d5aqoep, 2025-03-10:
I have kept my RM551E-GL in ECM mode. Internet works decent when directly connected to Windows PC's RJ45 port.

However if I reboot the RM551E-GL module, it creates network 2, 3, 4, .... and so on upon boot. Now I have reached Network 69 in Windows. I am already on the latest Dec 2024 firmware for the module. Is there any setting in OpenWRT which prevents these extra network creation ?

Another issue is when I feed the internet from M.2 to RJ45 enclosure into Ubiquiti Gateway, the gateway receives DHCP LAN IP 192.168.224.x and also gets working IPv6. But internet keeps on disconnecting/reconnecting every few minutes. Is it possible to create some kind of pure passthrough mode so that Ubiquiti is directly receiving the IP address from my 5G operator?

dr-dolomite, 2025-03-13:
The network profile issue is more of a Windows thing. You may stop this by setting a static IP for your PC through luci or through uci like this:

```
uci add dhcp host # =cfg0cfe63 (use your own host)
uci set dhcp.@host[-1].name='PC-NAME'
uci set dhcp.@host[-1].ip='192.168.227.79'
uci set dhcp.@host[-1].mac='84:A9:38:BF:D0:8F'
```

About the disconnection, lets wait @iamromulan if he got some ideas for it. However, if you want to use IP pass-through mode, it is available via Advance Connection settings.

[image]

clndwhr, 2025-03-25:
> The Ubiquiti problem persists. The 551 module cannot maintain stable connection when connected to Ubiquiti

Can you explain your network setup between the Ubiquiti gateway and the M.2 enclosure?  Asking as I'm not experiencing any issues with the 551E-GL module through my M.2 to RJ45 enclosure to my UDM-Pro.

MiG-41, 2025-03-25:
> as I'm unable to even get my 551 into IPPT mode

I noticed , that when you set  IP Pasthrough Mode and select MAC once , there is no possible to set another MAC , it simply don't change. It is neccesery to clean MPDN rules via AT+QMAP="MPDN_rule",0   and set AT+QMAPWAC=1 ( becouse it changes to 0 after cleaning) , and then it is possible to set it again to niew MAC...

clndwhr, 2025-03-25:
@MiG-41 I found the issue that was blocking me from being able to set into IPPT, The sdxpinn-mount-fix was at version 1.2.0 and it's blocking the QMAP command, updating to 1.3.2 allows me to input the command in, however as soon as i set AT+QMAP="IPPT_NAT",0 the entire device reboots, and somethign on my device keeps the mount-fix 1.3.2 from staying and it reverts back to 1.2.0 and removes all IPPT settings that were applied.

MiG-41, 2025-03-25:
But set AT+QMAP="IPPT_NAT",0 you will loose acces to toolkit...

clndwhr, 2025-03-25:
@MiG-41 Right, but having Tailscale or WG(if you can) resolves the access piece.  My concern more is setting IPPT_NAT,0 is resetting the device(i expect this) but none of the IPPT configurations are persisting

dr-dolomite, 2025-03-25:
This might be more of a firmware issue rather than our toolkit.

clndwhr, 2025-03-26:
You can't rule out the sdxpinn-mount-fix upgrade from 1.2.0 to 1.3.2 not persisting on a reboot regardless of executing any AT commands.

dr-dolomite, 2025-03-26:
> Generally , how to "clean" whole RM551E-GL ? By format /overlay ( just like in OpenWRT) , or flash fw via QFlash ? Or ?

Flash firmware via QFlash / QFirehose

clndwhr, 2025-03-27:
> Generally , how to "clean" whole RM551E-GL ? By format /overlay ( just like in OpenWRT) , or flash fw via QFlash ? Or ?

This resolved my IPPT issues AND the mount-fix upgrade issue

MiG-41, 2025-03-28:
Yes , that is emoving whole SDXPIN toolkit.
However things ,like acces to ADB ( no need to generate key) , preset of usbnet , AT+QMAPWAC set to eth, remains... 
Do you know how to reset whole RM551E-GL to "factory default".

dr-dolomite, 2025-03-28:
Probably flashing a clean QCN profile from a fresh factory modem might work

iamromulan, 2025-03-29:
- https://mega.nz/folder/CRFWlIpQ#grOByBgkfZe5uLMkX2M2XA 
- QCN profiles for the 551 are saved in the same place as firmware
- QPST under tools can be used to restore the qcn backup.
- Use the DM port, chose the software download utility, then go to the restore tab. Allow ESN mismatch.

d5aqoep, 2025-04-04:
> > The Ubiquiti problem persists. The 551 module cannot maintain stable connection when connected to Ubiquiti
> 
> Can you explain your network setup between the Ubiquiti gateway and the M.2 enclosure? Asking as I'm not experiencing any issues with the 551E-GL module through my M.2 to RJ45 enclosure to my UDM-Pro, but I am also going through the M.2 to RJ45 enclosure on a NAT configuration as I'm unable to even get my 551 into IPPT mode

I am using a random Aliexpress m.2 to rj45 enclosure to feed into Ubiquiti UCG-Fiber and IP passthrough mode enabled. It disconnects and reconnects every 5-10 mins.

dr-dolomite, 2025-04-04:
How many amperes does your power supply have?

clndwhr, 2025-04-04:
For reference I'm powering M.2 to RJ45 enclosure through a Ubiquiti PoE Injector. The distance from the injector to the enclosure is about 100ft on Cat6A UTP with 1 Cat6 female to female patch panel plug, and then another 6ft Cat6A to the enclosure; 3ft of cable from the injector to the UDM-Pro router.  I have also had it powered via a TPLink PoE unmanaged switch with no issues.  I did get IPPT working last week and had a stable connection, found zero noticeable differences on user experience, reverted back to NAT'd config yesterday for easier access to the modem itself without having to go through tailscale connection for other reasons. 

To note the RM551 x75 does require 12V 3Amp DC power.

@d5aqoep, Did you buy all the parts and put together and build the device or was it a purchase with all items included and pre-installed?  My UDM gets a 192.168.225.198/22 IPv4 from the OpenWrt lan-br in NAT'd mode. The OpenWrt lan-br interface being 192.168.224.0/22, usable IP range of 192.168.224.1 - 192.168.227.254, subnet being 255.255.252.0

@d5aqoep, whats the chance you think the latency monitoring tool on the Unifi OS is partly causing the disconnect? You  should be able to set your Internet Verification Server to your gateway's IPv4 address, your latency graph will constantly show 1ms because it's basically testing against itself, but this could be causing your provider to force disconnect under a presumption of an ICMP DDoS attack... Just a thought worth trying IMO. 

Image for reference on the Unifi OS, options through the Unifi Devices List, select your UCG and change settings.  

[image]

dr-dolomite, 2025-04-04:
> > How many amperes does your power supply have? Also, whats the response for AT+QTEMP?
> 
> 12V 2A power supply. I even tried another Type C adapter capable of doing 12V 3A. But that thing just keeps on disconnecting. Either I have a faulty enclosure or my 551E-GL is busted. I have ordered a V3 enclosure that is powered by PoE and will use the only PoE port of my UCG-Fiber to power it and then see. I also changed ping server to 1.1.1.1 but still disconnects.

You need to use the 12v barrel power input with at least 3A or 5A (recommended). Its type C USB port cannot provide enough power.

However, I think that the main culprit is your 5G to ETH board especially if its from MCUZone. If you can, just get the 5G2PHY adapted from rework network. We already tested it thoroughly and we had no issues with it.

d5aqoep, 2025-04-05:
> > > How many amperes does your power supply have? Also, whats the response for AT+QTEMP?
> > 
> > 
> > 12V 2A power supply. I even tried another Type C adapter capable of doing 12V 3A. But that thing just keeps on disconnecting. Either I have a faulty enclosure or my 551E-GL is busted. I have ordered a V3 enclosure that is powered by PoE and will use the only PoE port of my UCG-Fiber to power it and then see. I also changed ping server to 1.1.1.1 but still disconnects.
> 
> You need to use the 12v barrel power input with at least 3A or 5A (recommended). Its type C USB port cannot provide enough power.
> 
> However, I think that the main culprit is your 5G to ETH board especially if its from MCUZone. If you can, just get the 5G2PHY adapter from rework network. We already tested it thoroughly and we had no issues with it.

Yes mine is from MCUZone V2 version. I have ordered V3 PoE from same dealer. Hope it has fixes for stability issues. I don’t have 12V 3A adapter. That’s 36W. Does 551 module need that much?

[image]

dr-dolomite, 2025-04-05:
Mcuzone's boards cause disconnects with RM551E. The only board we could confidently recommend is the one from Rework

MiG-41, 2025-04-05:
Acc to my measurements RM551E-GL (together with Rework device) not need so much power , during UL max 6W ( however i don't have possibility to test with 5G aggregation  , only NSA n78...).  
But for sure power from USB is not enough ,and also experienced to random disconections ,when i used only USB cable. I belive 12V 3A is with large reserve ( not saying about 5A)...

More power consumption i experienced with RM520N-GL...

dr-dolomite, 2025-04-05:
I highly doubt that it is power related anymore. MCUZone's board were infamous with this problem even with RM520N-GL. The exact cause? I have no idea.

d5aqoep, 2025-04-08:
@dr-dolomite Hello. I am thinking of buying Rework Enclosure. Does it support 2.5G speeds?

MiG-41, 2025-04-08:
Yes , Rework Enclosure has 2,5G RJ45  ,and connection speed is 2500... ( another thing is , if someone exceed 1Gbps real speed from modem).

clndwhr, 2025-04-08:
> Yes , Rework Enclosure has 2,5G RJ45 ,and connection speed is 2500... ( another thing is , if someone exceed 1Gbps real speed from modem).

I regularly see 1.8Gbps, and I've seen up to max 2.5Gbps speeds on download.

MiG-41, 2025-04-08:
Had RM520N-GL with Rework , and there was not able to exceed 600Mbps via RJ45 ( but thru USB i get max from BTS , with was 950Mbps) .
That is why i'm little worries , if Ethernet is not something ,with limits....

clndwhr, 2025-04-08:
Wanted to add that I see the 1.8Gbps download speeds on my Ubiquiti UDM Pro, with the rework in NAT mode. Not just on the modem itself.

d5aqoep, 2025-04-13:
> > Yes , Rework Enclosure has 2,5G RJ45 ,and connection speed is 2500... ( another thing is , if someone exceed 1Gbps real speed from modem).
> 
> I regularly see 1.8Gbps, and I've seen up to max 2.5Gbps speeds on download.

I have never seen my RM551E-GL exceed 700 Mbps whereas my OnePlus 12 having same SD X75 modem easily got 1.2 Gbps at exact same location.

d5aqoep, 2025-04-15:
The issue is with Macuzone M.2 to RJ45 enclosure. Both V2 and V3 (PoE) model/designs are faulty which causes instability and disconnection issues with Ubiquiti. I switched to 5G Vehicle Router from Alibaba seller. Product name WS1610
https://www.alibaba.com/product-detail/Dual-band-3000Mbps-wifi6-5g-router-1601187160673.html
The support lady has been providing steady firmware updates since 6 months. I had it all along but had forgotten about it 2 months ago. So now switched to it. I also have an enclosure from rework on order. But I don't think I might need to use it lol.

closed 2025-05-15 by iamromulan