iamromulan/quectel-rgmii-toolkit #134: Issues trying to run the SDXPINN_toolkit.sh

Dmacmod, 2025-03-20:
Hi
When trying to follow the steps on the video to configure the Quectel RM551E-GL and running the command:
cd /tmp && wget -O rcPCIe_SDXPINN_toolkit.sh https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/rcPCIe_SDXPINN_toolkit.sh && chmod +x rcPCIe_SDXPINN_toolkit.sh && ./rcPCIe_SDXPINN_toolkit.sh && cd /

I get the following:
Downloading 'https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/rcPCIe_SDXPINN_toolkit.sh'
Failed to send request: Operation not permitted
/tmp #

MiG-41, 2025-03-21:
Do you have adb open and connected ?

dr-dolomite, 2025-03-31:
Will close this for now since no further response was made. Kindly read the repo documentation to properly install the toolkit. Feel free to open this issue again if the problem persists.

clndwhr, 2025-04-07:
@Dmacmod, I had this same issue and it was due to a bad flash of the firmware to the device. Try rerunning the flashing procedures with QFlash