iamromulan/quectel-rgmii-toolkit #136: Band Locking issue

STEVEBRZZW, 2025-03-22:
5g-sa band locking show blank, unable to select any band

[image]

MiG-41, 2025-03-22:
What model is this ? 
There is not so much bands ,with this modem support ( at least what  toolkit detect) , and looks like SA mode is not supported at all...

dr-dolomite, 2025-03-23:
We don't put static bands at all. What we queried from the modem will be the bands that will be shown. It seems that your modem doesn't support SA. However, to double check, maybe you can query this command and send the output here:

```
AT+QNWPREFCFG="policy_band"
```

dr-dolomite, 2025-03-24:
Closing this for now since its highly unlikely that its toolkit issue. I will open this again when I saw something wrong based on your ```AT+QNWPREFCFG="policy_band"``` result.

STEVEBRZZW, 2025-03-24:
[image]

MiG-41, 2025-03-24:
@STEVEBRZZW so what is the modem model ? can you also pass result of ATI command ?

dr-dolomite, 2025-03-25:
> [@STEVEBRZZW](https://github.com/STEVEBRZZW) so what is the modem model ? can you also pass result of ATI command and fw wersion AT+QGMR

@STEVEBRZZW Kindly provide the model name too.

STEVEBRZZW, 2025-03-27:
i am using a RM551E-GL

dr-dolomite, 2025-03-27:
> i am using a RM551E-GL

Try removing the QuecManager using Luci then install the latest beta

MiG-41, 2025-03-28:
Today i noticed, that also at my side it is like this:

[image]

And i'm not able to set any NSA bands...

MiG-41, 2025-03-28:
Maybe there is some exotic MBN Profile set ? ( I had such issue with RM530N-GL , when it was not able to connect at all , disabling Autoselect bring it to live).
@STEVEBRZZW ,can you show  Cellurar/Basic web page in QuecManager ?

dr-dolomite, 2025-03-31:
So far, I still cannot replicate this issue on my end.

STEVEBRZZW, 2025-04-01:
> Maybe there is some exotic MBN Profile set ? ( I had such issue with RM530N-GL , when it was not able to connect at all , disabling Autoselect bring it to live). [@STEVEBRZZW](https://github.com/STEVEBRZZW) ,can you show Cellurar/Basic web page in QuecManager ?

very helpful，i try to manually select a MBN Profile and all the NR-SA bands show up, the band locking is working now

closed 2025-04-01 by dr-dolomite