iamromulan/quectel-rgmii-toolkit #137: IPv4 Access from Modem with Passthrough?

jhenkens, 2025-03-23:
Hi,

When I enable IPPT via SimpleAdmin on the modem, I lose the ability to connect to IPv4 services from the modem itself.

```
root@sdxlemur:~# ping ipv6.google.com
PING ipv6.google.com(nuq04s45-in-x0e.1e100.net (2607:f8b0:4005:812::200e)) 56 data bytes
64 bytes from nuq04s45-in-x0e.1e100.net (2607:f8b0:4005:812::200e): icmp_seq=1 ttl=53 time=43.8 ms
64 bytes from nuq04s45-in-x0e.1e100.net (2607:f8b0:4005:812::200e): icmp_seq=2 ttl=53 time=42.0 ms
^C
--- ipv6.google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 41.972/42.873/43.775/0.901 ms
root@sdxlemur:~# ping google.com
PING google.com(sfo03s32-in-x0e.1e100.net (2607:f8b0:4005:814::200e)) 56 data bytes
64 bytes from sfo03s32-in-x0e.1e100.net (2607:f8b0:4005:814::200e): icmp_seq=1 ttl=53 time=49.9 ms
64 bytes from sfo03s32-in-x0e.1e100.net (2607:f8b0:4005:814::200e): icmp_seq=2 ttl=53 time=44.7 ms
^C
--- google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 44.738/47.330/49.922/2.592 ms
root@sdxlemur:~# ping ipv4.google.com
PING ipv4.l.google.com (142.251.46.238) 56(84) bytes of data.
ping: sendmsg: Operation not permitted
ping: sendmsg: Operation not permitted
^C
```
It appears to work correctly at my router/laptop level, going through the modem, just not connections originating on the modem itself (thus, I can't speedtest, or download the latest version of the toolkit, without disabling IPPT).

iamromulan, 2025-03-23:
Which module is it? I'll try to see if I see the same thing.

ripclap, 2025-04-10:
**Modem**: RM520N-GL  
**Firmware Version**: RM520NGLAAR01A08M4G_01.206.01.206  

**Summary:**  
The modem's rmnet_data1 interface (IP 169.254.5.1) wasn't accessing the Internet because an iptables rule was dropping all traffic from the 169.254.0.0/16 range. 

This rule is created automatically when running: 
```AT+QMAP="mpdn_rule",0,1,0,1,1,"FF:FF:FF:FF:FF:FF"``` 

This issue was replicated on firmware version RM520NGLAAR01A08M4G_01.206.01.206 and does not occur on other releases. 

**Workaround:**  

1. **Find the rule:**  
   Run this command to list the OUTPUT chain with line numbers:
   ```bash
   iptables -L OUTPUT --line-numbers
   ```
   You should see something like:
   ```
   Chain OUTPUT (policy ACCEPT)
   num  target     prot opt source               destination
   1    DROP       all  --  169.254.0.0/16       anywhere
   ```

2. **Delete the rule:**  
   Remove the rule by its number (for example, if it's rule number 1):
   ```bash
   iptables -D OUTPUT 1
   ```

After that, the modem's internal connection should work fine.

jhenkens, 2025-04-10:
Sorry, somehow I missed the initial reply in my inbox. Yes, RM520N.

ripclap, 2025-04-12:
@jhenkens

Send the output of:
AT+QGMR

jhenkens, 2025-04-14:
```
AT+QGMR

RM520NGLAAR01A08M4G_01.206.01.206

OK
```

ripclap, 2025-04-15:
> ```
> AT+QGMR
> 
> RM520NGLAAR01A08M4G_01.206.01.206
> 
> OK
> ```

I shared a workaround above, but if you don't specifically need that firmware version, I recommend downgrading — the issue doesn’t occur in earlier versions.

closed 2025-05-15 by iamromulan