iamromulan/quectel-rgmii-toolkit #140: Wireguard Support

ripclap, 2025-03-25:
Have you had any success getting Wireguard/Wireguard-go to work? I understand that Tailscale is a fork of wg-go, so there might be some potential there.

MiG-41, 2025-03-25:
There is a package wireguard-tools ,with should be enough to work, why don't you try ?

nonameirl, 2025-03-25:
Kernel 5.4 does not include the wireguard module. Wireguard-go is the userland workaround. Nether work as is.

Have you tried it?

> There is a package wireguard-tools ,with should be enough to work, why don't you try ?

MiG-41, 2025-03-25:
No , i don't use wireguard....  But just from this description i assume would be enough: https://eko.one.pl/?p=openwrt-wireguard

ripclap, 2025-03-25:
> No , i don't use wireguard.... But just from this description i assume would be enough: https://eko.one.pl/?p=openwrt-wireguard

It's not enough. Which is why I'm hoping someone a part of the project can fill me in.

MiG-41, 2025-03-26:
If you say so. 
However i tried Zerotier , and works.

dr-dolomite, 2025-03-26:
Some already tested WG using Cloudflare Warp config and it works. It might get added in the future.

ripclap, 2025-03-26:
> Some already tested WG using Cloudflare Warp config and it works. It might get added in the future.

Do you have a link to that discussion?

dr-dolomite, 2025-03-26:
It was done through private testing to see if its feasible. @iamromulan knows more about this topic.

nonameirl, 2025-03-27:
> It was done through private testing to see if its feasible. @iamromulan knows more about this topic. 

If there's any way I could get access to that, even just for testing purposes, it would be incredibly helpful. I’m working against a deadline, and my own attempts at testing and compiling haven’t been successful so far—having something to reference or build upon would really make a difference.

ripclap, 2025-03-27:
> It was done through private testing to see if its feasible. @iamromulan knows more about this topic. 

If there's any way I could get access to that, even just for testing purposes, it would be incredibly helpful. I’m working against a deadline, and my own attempts at testing and compiling haven’t been successful so far—having something to reference or build upon would really make a difference.

iamromulan, 2025-03-27:
Apologies. I've been pretty short on time as of late. 

Wireguard is only available for SDXPINN right now and is experimental. Warp is installable from Luci software. It will install the decencies along with it including the wireguard Luci app I jacked from Rooter. You'll notice that there is a kmod-wireguard package as well. This is an empty package and only installed in order to register with opkg that the dependency for it is satisfied. Wireguard is already compiled into the kernel. We see this by running the dmesg kernel log as soon as possible after boot. If you scroll all the way to the top and then start scrolling down you'll eventually find a line that mentions wireguard. I have not confirmed that this is also in existence on sdxlemur. Nonetheless, sdxlemur is like a neglected child right now for me. Well little time I have had lately has been going to SDXPINN, which is why it's a great idea to acquire an RM551E-GL if you haven't already. 

Anyways, after installing warp, head over to the terminal. Type warp free setup

After that log out login to Luci and you should see a pre-configured VPN profile you can test in the Luci app. 

Click start, but know that it probably will not work right away. You will need to head over to firewall and add wg0 as a member interface of the WAN zone. It's important to note that masquerading is required for this to work so definitely keep that checked. 

Then head back to the wireguard Luci app and stop / start the warp VPN profile. From there things should be working and you will be free to add your own profiles.

ripclap, 2025-03-28:
@iamromulan Thanks for the reply. It would be easier to upgrade the device for sure.

Here’s your revised GitHub comment with properly formatted command output using GitHub’s Markdown syntax:  

---

**Great news! I got it working today! 🎉**  

Here’s what I did:  

1. Installed the necessary Entware packages: `wireguard-tools` and `wireguard-go`.  
2. Removed the existing `wg` binary from `/opt/bin/`.  
3. Compiled [`[seud0nym/wg-go](https://github.com/seud0nym/wg-go)`](https://github.com/seud0nym/wg-go) for ARM.  
4. Copied the compiled binary to `/opt/bin/wg` and created a symlink:  
   ```sh
   ln -s /opt/bin/wg /bin/wg
   ```
5. Updated my WireGuard config (`/etc/wireguard/wg0.conf`):  
   - Removed the `Address = x.x.x.x/x` line from `[Interface]`.  
   - Added a `PostUp` rule:  
     ```ini
     PostUp = ip address add x.x.x.x dev wg0
     ```
6. Started WireGuard:  
   ```sh
   wg-quick up wg0
   ```  

### Command Output:
```sh
root@sdxlemur:~# wg-quick up wg0
[#] ip link add wg0 type wireguard
Error: Unknown device type.
[!] Missing WireGuard kernel module. Falling back to slow userspace implementation.
[#] wireguard-go wg0
┌──────────────────────────────────────────────────────┐
│                                                      │
│   Running wireguard-go is not required because this  │
│   kernel has first class support for WireGuard. For  │
│   information on installing the kernel module,       │
│   please visit:                                      │
│         https://www.wireguard.com/install/           │
│                                                      │
└──────────────────────────────────────────────────────┘
[#] wg setconf wg0 /proc/self/fd/63
2025/03/28 07:19:03 Interface wg0 configured.
[#] ip link set mtu 1350 up dev wg0
[#] ip -4 route add 172.16.0.0/16 dev wg0
[#] ip address add 172.16.32.12 dev wg0

root@sdxlemur:~# wg show
Interface: wg0 (userspace)
  public key: tP3hJ9v1aKq6yL0X+ZsQ9vG5N4Tz8JY0hL2V2X5aW0Y=
  private key: (hidden)
  listening port: 34710

peer: W4x9V7rPzQK8yL3G+N5Tq2JX0hV6aZ8W1Y9L0X5K0Q=
  endpoint = 192.83.147.215:51820
  allowed ips = 172.16.0.0/16
  keep alive interval = 120s
  last handshake time = 2025-03-28T07:19:03Z
  transfer: 188 bytes received, 276 bytes sent
  protocol version = 1
```