iamromulan/quectel-rgmii-toolkit #142: SCC newline delineation missing with extended NR5G Extended Signal from CAINFO

clndwhr, 2025-03-29:
On QuecManager 2.1.5 Beta. When setting `AT+QCAINFO=1`, the fetch_data.sh?set=1 response array index 13, referencing +QCAINFO query, is missing the newline delimiter between SCC responses before the second SCC +QCAINFO entry.

`{
    "command": "AT+QCAINFO",
    "response": "AT+QCAINFO\n+QCAINFO: \"PCC\",501390,12,\"NR5G BAND 41\",153,-11,-11,2549\n+QCAINFO: \"SCC\",126270,2,\"NR5G BAND 71\",1,93,0,-,-+QCAINFO: \"SCC\",521310,11,\"NR5G BAND 41\",1,153,0,-,-,-11,-11,2549\n",
    "status": "success"
}`

Returning to `AT+QCAINFO=0` the response contains the correctly expected newline delimiter.  I don't believe this to be an issue in how fetch_data.sh is parsing the response, but more the response from the QCAINFO query does not contain the expected format to parse for all info, missing the newline delimiter.

dr-dolomite, 2025-03-29:
We are already aware of this but sadly Quectel still didn't fix how QCAINFO works for RM551E. It doesn't work or behave as intended.

MiG-41, 2025-03-30:
And now what is the difference beetween 1 and 0 ? In LTE I dont see any... why to set it to 1 ? (Default is 0 ?)

dr-dolomite, 2025-03-30:
> And now what is the difference beetween 1 and 0 ? In LTE I dont see any... why to set it to 1 ? (Default is 0 ?)

Should be for enabling verbosity to QCAINFO result

MiG-41, 2025-03-30:
Verbosity in a way of more frequent change , or more parameters displayd ? ( since i havn't noticed difference... in what is displayd , same parameters in LTE bands).

dr-dolomite, 2025-03-30:
"Shoul be" more parameters but then again, QCAINFO doesn't properly work on RM551E right now.

MiG-41, 2025-03-30:
Ok , in NSA i see the difference:

```
AT+QCAINFO
+QCAINFO: "PCC",225,75,"LTE BAND 1",1,234,-91,-8,-65,9
+QCAINFO: "SCC",1599,50,"LTE BAND 3",2,423,-94,-11,-66,3,0,-,-
+QCAINFO: "SCC",647424,12,"NR5G BAND 78",647
OK
```

and with 1:

```
AT+QCAINFO
+QCAINFO: "PCC",225,75,"LTE BAND 1",1,234,-92,-10,-64,5
+QCAINFO: "SCC",1599,50,"LTE BAND 3",2,423,-94,-13,-62,0,0,-,-
+QCAINFO: "SCC",647424,12,"NR5G BAND 78",647,-12,-11,526
OK
```

clndwhr, 2025-03-30:
> "Should be" more parameters but then again, QCAINFO doesn't properly work on RM551E right now.

It is working properly, the formatting isn't what is expected.  Why not split the response on `+QCAINFO` instead of `\n`, this supports both non-verbose and verbose output for NR5G bands and LTE Bands combined.

dr-dolomite, 2025-03-31:
Actually, theres no problem with NSA and LTE since Im using servingcell for NR5G band. This combination could properly display the needed values without changing the default behavior of QCAINFO. However, the SCC bands of NR5G-SA doesnt return the expected values even with verbosity activated (QCAINFO=1) as shown here:

```
+QCAINFO: "PCC",<freq>,<NR_DL_bandwidth>,<NR_ba
nd>,<PCID>[,<NR_RSRP>,<NR_RSRQ>[,<NR_SNR>]]
[+QCAINFO: "SCC",<freq>,<NR_DL_bandwidth>,<NR_b
and>,<scell_state>,<PCID>,<UL_configured>,<NR_UL_b
andwidth>,<UL_ARFCN>[,<NR_RSRP>,<NR_RSRQ>[,<N
R_SNR>]]
[…]
```

Thats why I said that QCAINFO on RM551E is still bugged. Sorry but no plans to change the QCAINFO behavior for now since it doesnt really help much. I also don't have a copy of NSA with 2 SCC NR5G bands so no idea how servingcell and QCAINFO responses would be shown.

dr-dolomite, 2025-03-31:
Even at this part, it doesnt show the expected response.

> ```
> AT+QCAINFO
> +QCAINFO: "PCC",225,75,"LTE BAND 1",1,234,-92,-10,-64,5
> +QCAINFO: "SCC",1599,50,"LTE BAND 3",2,423,-94,-13,-62,0,0,-,-
> +QCAINFO: "SCC",647424,12,"NR5G BAND 78",647,-12,-11,526
> OK
> ```

From RM551E document:
```
In EN-DC mode:
+QCAINFO: "PCC",<freq>,<bandwidth>,<band>,<pcell_s
tate>,<PCID>,<RSRP>,<RSRQ>,<RSSI>,<RSSNR>
[+QCAINFO: "SCC",<freq>,<bandwidth>,<band>,<scell_
state>,<PCID>,<RSRP>,<RSRQ>,<RSSI>,<RSSNR><UL_
configured>,<UL_bandwidth>,<UL_EARFCN>]
[…]
[+QCAINFO: "SCC",<freq>,<NR_DL_bandwidth>,<NR_b
and>,<PCID>][,<NR_RSRP>,<NR_RSRQ>[,<NR_SNR>]]
[+QCAINFO: "SCC",<freq>,<NR_DL_bandwidth>,<NR_b
and>,<scell_state>,<PCID>,<UL_configured>,<NR_UL_b
andwidth>,<UL_ARFCN>[,<NR_RSRP>,<NR_RSRQ>[,<NR_SNR>]]
```

iamromulan, 2025-03-31:
With time comes progress 

I heard a rumor that 
``RM551EGL00AAR01A03M8G_A0.001.A0.001`` is coming very very soon.

Hopefully it will be fixed.

dr-dolomite, 2025-03-31:
The new firmware fixed some of the AT+QCAINFO's bugged response data. Will close this for now as I will rewrite the needed parsing logic to fit the new firmware as well as add support for older one.

MiG-41, 2025-03-31:
> With time comes progress
> 
> I heard a rumor that `RM551EGL00AAR01A03M8G_A0.001.A0.001` is coming very very soon.
> 
> Hopefully it will be fixed.

And it is for sure realesed yesterday.

MiG-41, 2025-03-31:
In AT+QENG="servingcell" thera are all parameters. But not sure if it is presents only one band for 5G...

And there are also commands: AT+QRSRP AT+QRSRQ AT+QSINR ....

clndwhr, 2025-03-31:
> Actually, theres no problem with NSA and LTE since Im using servingcell for NR5G band. This combination could properly display the needed values without changing the default behavior of QCAINFO. [...]

For your reference, NSA Mode with 2 LTE/2NR5G 

`{
    "command": "AT+QCAINFO",
    "response": "AT+QCAINFO\n+QCAINFO: \"PCC\",854,50,\"LTE BAND 2\",1,126,-93,-9,-66,12\n+QCAINFO: \"SCC\",66736,50,\"LTE BAND 66\",1,15,-89,-9,-72,17,0,-,-\n+QCAINFO: \"SCC\",501390,12,\"NR5G BAND 41\",153,-11,-11,2378\n+QCAINFO: \"SCC\",521310,11,\"NR5G BAND 41\",1,153,0,-,-,-11,-11,2378\n",
    "status": "success"
}`

MiG-41, 2025-03-31:
In new fw there is mentioned:
`Secure Login AT+QADBKEY was no longer supported`

Not sure if this couse ,that there will be no possible to wget toolkit via adb...

dr-dolomite, 2025-03-31:
"Shoul be" more parameters but then again, QCAINFO doesn't properly work on RM551E right now.

We are already using the latest firmware. Just make sure that you have unlocked adb before flashing this version.

STEVEBRZZW, 2025-04-01:
@iamromulan @dr-dolomite 
wander where can i get the a03 firmware 

also https://forums.quectel.com/t/requesting-rm551egl00aar01a03m8g-a0-001-a0-001/45554/6
i see romulan saying it's worse than the a02 firmware, why say so?

dr-dolomite, 2025-04-01:
> @iamromulan @dr-dolomite 
> wander where can i get the a03 firmware 
> 
> also https://forums.quectel.com/t/requesting-rm551egl00aar01a03m8g-a0-001-a0-001/45554/6
> i see romulan saying it's worse than the a02 firmware, why say so?

https://mega.nz/folder/CRFWlIpQ#grOByBgkfZe5uLMkX2M2XA/file/3d9EVbYR

New firmware removed the 4CC for NR5G-SA. It also removed the AT commands for unlocking ADB.

STEVEBRZZW, 2025-04-01:
> https://mega.nz/folder/CRFWlIpQ#grOByBgkfZe5uLMkX2M2XA/file/3d9EVbYR
>
> New firmware removed the 4CC for NR5G-SA. It also removed the AT commands for unlocking ADB.

4CC NR5G-SA isn't my concern since my carrier doesn't support it.
And i have already unlocked ADB.

i am currently running the a01 firmware, would you recommend updating to a02 or a03?

dr-dolomite, 2025-04-01:
You could test it out and personally see if there are improvements on your end. The toolkit and QuecManager is compatible with the latest firmware.

STEVEBRZZW, 2025-04-01:
> You could test it out and personally see if there are improvements on your end. The toolkit and QuecManager is compatible with the latest firmware.

OK
Thanks a lot.
In fact i am using some tricks to convert a Fibcom FM190W-GL into RM551E-GL, wander if i can follow the normal update procedure designed for a real RM551E-GL. Or do i need to flash the firmware and the xqcn file.

https://github.com/doner005/RMFaux51/blob/master/RMFaux51.md
This is the guide i follow

clndwhr, 2025-04-03:
> https://mega.nz/folder/CRFWlIpQ#grOByBgkfZe5uLMkX2M2XA/file/3d9EVbYR
>
> New firmware removed the 4CC for NR5G-SA. It also removed the AT commands for unlocking ADB.

Is there a chance for the 4CC fix with the new firmware?

dr-dolomite, 2025-04-03:
Its a Quectel issue so theres nothing we can do about it rather than to wait for another firmware or hope that someone leaks a proprietary fix.