iamromulan/quectel-rgmii-toolkit #147: Cell Scanner Bugs

STEVEBRZZW, 2025-04-09:
[image]
When scanning nr5g cells, bandwidth and cell id wont show correctly

[image]
under detail information, scs wont show

and after i update to the 2.2.3 quecmanager, the ‘Clear Result’ button stopped working. i click it and result is cleared, refresh the page, results are back.

wondering if it's a quecmanager bug or just quectel's AT command being buggy again.

dr-dolomite, 2025-04-09:
What firmware version?

STEVEBRZZW, 2025-04-09:
r01a03

clndwhr, 2025-04-09:
@dr-dolomite, I'm seeing the same issue with PCI and the Clear Results issue. Here's my NR output for reference:

```
{
    "status": "success",
    "timestamp": "2025-04-09 13:43:35",
    "output": "AT+QSCAN=3,1\n+QSCAN: \\\"NR5G\\\",310,260,501390,153,-76,-11,33,1,105FB012D,A80000,273,41,36,0,-\n+QSCAN: \\\"NR5G\\\",310,260,126270,93,-80,-11,37,0,105FB0001,A80000,79,71,12,0,-\n+QSCAN: \\\"NR5G\\\",310,260,393850,696,-89,-11,30,0,105FB0015,A80000,106,25,38,10,-\n+QSCAN: \\\"NR5G\\\",310,260,521310,153,-78,-11,31,1,105FB0137,A80000,245,41,34,14,-\n+QSCAN: \\\"NR5G\\\",313,340,401050,44,-118,-13,3,0,03C4E7077,6143,133,70,44,0,-\n+QSCAN: \\\"NR5G\\\",313,340,431530,44,-119,-14,2,0,03C4E707A,6143,25,66,3,4,-\n+QSCAN: \\\"NR5G\\\",313,340,123870,44,-102,-15,19,0,03C4E7071,6143,25,71,1,8,-\n+QSCAN: \\\"NR5G\\\",311,480,648672,20,-125,-17,-18,1,8EB10403A,E90114,273,77,140,6,-"
}
```

[image]

[image]

[image]

clndwhr, 2025-04-14:
@STEVEBRZZW, pushed a code fix for the Display parsing. This does not address the results returning after clicking the clear button

https://github.com/dr-dolomite/QuecManager-JS/pull/11

clndwhr, 2025-04-16:
@STEVEBRZZW, one more PR to address the NaN values (SCS may not be returning from the AT+QSCAN query), Unknown Bands in the Cell Freq Analysis section, and the Clear Results button not clearing the previous results from the persistent store on the modem.

@dr-dolomite 

https://github.com/dr-dolomite/QuecManager-JS/pull/12

closed 2025-05-15 by iamromulan