iamromulan/quectel-rgmii-toolkit #151: RM551E 2.5G network fail

STEVEBRZZW, 2025-04-21:
Mon Apr 21 17:59:40 2025 user.err AGM: device[14142]: device_init: 568 no valid snd device found retry 13 times
Mon Apr 21 17:59:41 2025 kern.err kernel: [  390.937729][   T37] eth0: pci link is down
Mon Apr 21 17:59:41 2025 kern.err kernel: [  390.937729][   T37] .
Mon Apr 21 17:59:41 2025 kern.err kernel: [  392.946454][   T37] eth0: cmd = 0xff, should be 0x06
Mon Apr 21 17:59:41 2025 kern.err kernel: [  392.946454][   T37] .
Mon Apr 21 17:59:43 2025 kern.err kernel: [  392.953170][   T37] eth0: pci link is down
Mon Apr 21 17:59:43 2025 kern.err kernel: [  392.953170][   T37] .
Mon Apr 21 17:59:43 2025 kern.err kernel: [  394.962001][   T37] eth0: cmd = 0xff, should be 0x06
Mon Apr 21 17:59:43 2025 kern.err kernel: [  394.962001][   T37] .
Mon Apr 21 17:59:45 2025 user.err AGM: device[14142]: device_init: 568 no valid snd device found retry 12 times
Mon Apr 21 17:59:45 2025 kern.err kernel: [  394.968680][   T37] eth0: pci link is down
Mon Apr 21 17:59:45 2025 kern.err kernel: [  394.968680][   T37] .
Mon Apr 21 17:59:45 2025 kern.err kernel: [  396.977826][   T37] eth0: cmd = 0xff, should be 0x06
Mon Apr 21 17:59:45 2025 kern.err kernel: [  396.977826][   T37] .
Mon Apr 21 17:59:47 2025 kern.err kernel: [  396.984695][   T37] eth0: pci link is down
Mon Apr 21 17:59:47 2025 kern.err kernel: [  396.984695][   T37] .
Mon Apr 21 17:59:47 2025 kern.err kernel: [  398.993504][   T37] eth0: cmd = 0xff, should be 0x06
Mon Apr 21 17:59:47 2025 kern.err kernel: [  398.993504][   T37] .
Mon Apr 21 17:59:49 2025 kern.err kernel: [  399.000342][   T37] eth0: pci link is down
Mon Apr 21 17:59:49 2025 kern.err kernel: [  399.000342][   T37] .
Mon Apr 21 17:59:49 2025 kern.err kernel: [  401.009303][   T37] eth0: cmd = 0xff, should be 0x06
Mon Apr 21 17:59:49 2025 kern.err kernel: [  401.009303][   T37] .

i use my rm551 to 2.5g board and connected to a travel router.
the network will work for around 1 minute, and it will cut out, but it only happens on the ethernet port. If i adb into the module, it does have network.
Is there any solution?

clndwhr, 2025-04-24:
@STEVEBRZZW,

What m.2 to RJ45 module are you using? What rm551 firmware do you have loaded?

iamromulan, 2025-04-25:
> using one i bought online, no branding or anything 
> I am currently running r01a03 firmware
> 
> > [@STEVEBRZZW](https://github.com/STEVEBRZZW),
> > 
> > What m.2 to RJ45 module are you using? What rm551 firmware do you have loaded?
> 
> 

Got a link to what you got?
Not all eth sleds are created equal. Some have issues with the PCIe link between the Ethernet chipset and modem.

STEVEBRZZW, 2025-04-25:
It used to work fine.
I wonder if it has anything to do with the a
03 firmware

iamromulan, 2025-04-25:
> It used to work fine.
> I wonder if it has anything to do with the a
> 03 firmware 

Oh you're on March firmware? Yeah no definitely get rid of that.
You can use my custom firmware (December firmware base) to skip on past ADB. First boot it send some at commands and reboots. Be sure to read my note

https://mega.nz/file/GUkk0RpD#7JsFymzKBBKplxLs5tqc5-m7rBmrB2QPfTfMtf1ZQ4g

d5aqoep, 2025-04-25:
> Oh you're on March firmware? Yeah no definitely get rid of that. You can use my custom firmware (December firmware base) to skip on past ADB. First boot it send some at commands and reboots. Be sure [...]

What is in your custom firmware?

iamromulan, 2025-04-25:
> What is in your custom firmware?

 -----------------------------------------------------
 Base firmware: RM551EGL00AAR01A02M8G_2024_12_23
 -----------------------------------------------------
 iamromulan - mod - basic_eth - 4/21/2025 - Change log:
 -----------------------------------------------------
 Target: Quectel RM551EGL00AA
 Branch: Basic_Eth 
 Description: Basic modifications and optimizations meant for use over direct Ethernet
--------------------------------------------------------------------------------
User space Login preset:
Username: root
Password: iamromulan
--------------------------------------------------------------------------------
This firmware is equipped with first boot AT Command execution. After flashing, keep it powered until it reboots one more time.
After that reboot, if you are using an Ethernet sled, Ethernet should be available.
Visit 192.168.224.1 in your web browser to access OpenWRT Luci
--------------------------------------------------------------------------------
First boot AT commands:
Enables/sets ECM USB mode. Ensures auto connect takes effect.
`AT+QCFG="usbnet",1`
Enables PCIe Root Complex mode. Enables use of an Ethernet chipset/port directly with the modem.
`AT+QCFG="pcie/mode",1`
Reboot
`AT+CFUN=1,1`
--------------------------------------------------------------------------------
Images modified:

# sysfs.ubi (rootfs which is mounted at / from Base firmware)

- Installed my mount-fix in order to create a manageable user space. This causes the original rootfs mounted at / to move to /real_rootfs 
- Re-added previously removed dropbear and uhttpd startup scripts and enabled both
	- As a result the OpenWRT Luci web interface will be available for use after flash
	- SSH is available for use after flash
- Installed sdxpinn-patch 2.5 which includes the following fixes:
	- Patched opkg distfeeds by commenting out invalid sources
	- Ensured the following sources were listed in distfeeds:
			- https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/base
			- https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/luci
			- https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/packages
			- https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/routing
			- https://downloads.openwrt.org/releases/22.03.5/packages/aarch64_cortex-a53/telephony/
	- Ensured my own repo was listed in custom feeds (https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/SDXPINN/opkg-feed)
	- Ensured my own repo's public key was installed
	- Patched the arch opkg config to reflect: arch all 1; arch noarch 1; arch sdxpinn 2; arch aarch64_cortex-a53 10; arch aarch64_generic 20
- Renamed the Quectel login binary from login to login.old (/bin/login.old) and replaced with the standard OpenWRT login utility from shadow-login.
- Saved this readme to /release.iamromulan.txt

# usrdata.ubi (usrdata which is mounted at /usrdata)

- The following packages and edits have been pre-installed to the stock usrdata image. It is predicted that they will be lost when running AT+QCFG="reset_factory"
- Packages:
	- atinout (basic in/out AT command utility)
	- luci-app-atinout-mod (luci app: for sending AT commands from the OpenWRT Luci Web interface. Includes a few saved commands ready for use.)
	- sdxpinn-console-menu (A work in progress basic number/selection based CLI for general management)
	- luci-app-ttyd (luci app: For accessing the shell/command line interface via the OpenWRT Luci Web interface.)
	- mc and mc-skins (A classic and solid CLI File Browser)
- Edits:
	- Saved this readme to /usrdata/release.iamromulan.txt
	
	
# NON-HLOS.ubi (modem which is mounted at /firmware)

 - Saved this readme to /firmware/release.iamromulan.txt

d5aqoep, 2025-04-26:
Installed. Working fine and seems to be more stable compared to A03

Will our Engineering Sample RM551E-GL become final versions with launch of General Availability or Retail units?

iamromulan, 2025-05-15:
> Installed. Working fine and seems to be more stable compared to A03
> 
> Will our Engineering Sample RM551E-GL become final versions with launch of General Availability or Retail units?

No idea. Seems like it since Fibocom's FM190W-GL is the same hardware.

closed 2025-05-15 by iamromulan