iamromulan/quectel-rgmii-toolkit #154: Independent webui

Sunser, 2025-04-30:
Can you consider separating the webui? In the module, you only need to install a socat to convert the at port to network at, and the webui is called through the network. This will first reduce the modification of the module, and secondly, the independent web is convenient for upgrading. For example, the web is packaged with docker, which is particularly convenient for upgrading.

Sunser, 2025-04-30:
Also, for example, when USB is connected to Linux, at operations can be performed through ttyUSBX

clndwhr, 2025-05-01:
@Sunser,

Please provide which modem and web UI you're requesting in a docker package.  

I wouldn't bet on the separation to occur for QuecManager as the QM package contains both "modem" scripts as well the client interface. Building this into a docker package would require additional separation of packages and more development and integration workload on the already few individuals that are executing development on this entire package.

You are more than welcome to fork the web UI and containerize that for your personal needs, just understand that features released in the web UI are also generally dependent on the logic and scripts available on the modem itself.

Sunser, 2025-05-01:
I am currently using RM520N-GL, https://hub.docker.com/r/huangfugui/mt5700-at This is an external webUI that I saw used by another module. Can you consider implementing such an external webUI

clndwhr, 2025-05-01:
@Sunser,

I'll have a docker container build pushed for the web UI over the weekend sometime at the earliest.  I will need feedback as I do not have the current hardware to work with that web UI specifically.

Sunser, 2025-05-01:
OK, thank you. If you need hardware debugging, you can contact me via WeChat. I provide AT port forwarding based on socat. My WeChat ID: odidcc. Download link for WeChat international version: https://www.wechat.com/en/
In addition, due to translation issues, some expressions may not be translated accurately, please forgive me.

clndwhr, 2025-05-01:
@Sunser, unfortunately I'm unable to utilize WeChat as a condition of my employment.  I'll look for feedback on here though.

Sunser, 2025-05-01:
Ok, if there is any other version besides docker or standalone webui I will test the functionality and give feedback whether it is available and if there are any problems

clndwhr, 2025-05-01:
https://hub.docker.com/r/clndwhr/rgmii-simpleadmin

Sunser, 2025-05-01:
Yes, it is not available at the moment. Are you considering implementing it directly based on QuecManager? Isn't simpleadmin going to be deprecated and not maintained?

clndwhr, 2025-05-01:
@Sunser 

tag 0.0.2 Should be mostly functional for you now.  When we get to the point to getting QuecManager handling both PINN and LEMUR systems we will work to get some builds in place in docker.

Sunser, 2025-05-01:
OK, I will take the time to test version 0.0.2 in the next few days and give feedback on whether it is usable.

clndwhr, 2025-05-01:
@Sunser 

Work with 0.0.3 or latest,  I was corrected in a network difference between lemur/prarie and PINN for the modem IP

Sunser, 2025-05-01:
I just pulled and looked at /etc/nginx/conf.d/default.conf , where /cgi-bin/ is redirected to http://192.168.225.1/cgi-bin/ . Without using adb to install the plugin, this URL is inaccessible. Shouldn't it be directly based on /dev/ttyUSBX or ip:port tcp access?

clndwhr, 2025-05-01:
This container build is 100% assuming you're using the RM500/502/520/521/530, with the toolkit installed, and to remove the front end processing from the modem itself.

It will take some time and effort to change the codebase to allow remote and local at commands in the manner you're asking for. I'm don't have the answer to where this would be in priority of development.  We'll keep this in mind as we develop the platforms going forward.

Sunser, 2025-05-01:
If you can't execute remote or local AT commands, the external webui is actually meaningless. You have to flash the toolkit, so the web external is meaningless. I will pay attention to your subsequent versions and hope that you will consider compatibility with the calling methods I mentioned.

clndwhr, 2025-05-01:
@Sunser, understandable your frustration with what was provided.  Out of the box you have to, at a minimum regardless of flashing the firmware or otherwise, connect via non-network to the device to configure the modem to at least put it into pcie/mode and/or usbnet mode, and still continue on to configure the socat device bridge, etc...  If you're looking for AT over IP:PORT in the current build you're more than welcome to fork the repo and work a solution and present a pull request.