iamromulan/quectel-rgmii-toolkit #162: SimpleAdminRev-Alpha-0.9 - wrong NR FDD (n28) bandwidth

1alessandro1, 2025-05-10:
Hello, in italy we only have 10MHz n28 available to customers. How does the program compute the bandwidth? Does it use the 3gpp formula from EARFCN?

```
AT+QCAINFO
+QCAINFO: "PCC",3175,75,"LTE BAND 7",1,154,-95,-11,-65,22
+QCAINFO: "SCC",1350,100,"LTE BAND 3",2,18,-92,-17,-52,17,0,-,-
+QCAINFO: "SCC",275,75,"LTE BAND 1",2,368,-92,-11,-62,24,0,-,-
+QCAINFO: "SCC",154570,1,"NR5G BAND 28",152

```

[image]

MiG-41, 2025-05-10:
Generally this is the case for n28 line:
`[+QCAINFO: "SCC",<freq>,<NR_DL_bandwidth>,<NR_band>,<PCID>]
`
and 1 in NR_DL_bandwitch means 10MHz... so something with parsing ( 55Mhz is not possible for bandwitch).

clndwhr, 2025-05-15:
@1alessandro1   

@MiG-41 

I have a bug fix submitted in to resolve this and is based on the same logic being used in QuecManager

iamromulan, 2025-05-17:
#163 has been merged to development-SDXLEMUR
Let me make sure the URL's in the deployment script are correct then I will close

iamromulan, 2025-05-17:
Ok deployment scripts are updated. Please give it a try, as my RM520 is not a available for testing at the moment. Installing simpleadmin over an existing install should be fine as well as the simpleadmin dir is removed in the process. 

Run this in adb sdxlemur shell or ttyd: "Console" from simpleadmin and install simpleadmin to test. If you run this from ttyd/"Console" from simpleadmin note that you will loose connection to the Console session when ttyd is being installed again. Allow the modem to run for a few min then check again. Even though you loose connection systemd continues the process separate from ttyd for this scenario. 

``adb shell "cd /tmp && wget -O RMxxx_rgmii_toolkit.sh https://raw.githubusercontent.com/iamromulan/quectel-rgmii-toolkit/development-SDXLEMUR/RMxxx_rgmii_toolkit.sh && chmod +x RMxxx_rgmii_toolkit.sh && ./RMxxx_rgmii_toolkit.sh" && cd /``

1alessandro1, 2025-05-17:
I confirm that the session will be closed upon updating simpleadmin: 

[image]

The latest commit has fixed the issue, now the bandwidth is evaluated correctly. Thank you @clndwhr @iamromulan 
[image]

closed 2025-05-17 by 1alessandro1

clndwhr, 2025-05-17:
@1alessandro1, can you confirm your PCI values as well?  The update includes a change in the logic for those also.

1alessandro1, 2025-05-17:
[image]

Values are correctly matching in order, yes

```
root@sdxprairie:~# atcmd at+qcainfo
at+qcainfo
+QCAINFO: "PCC",3175,75,"LTE BAND 7",1,154,-95,-6,-70,20
+QCAINFO: "SCC",275,75,"LTE BAND 1",1,368,-91,-8,-66,25,0,-,-
+QCAINFO: "SCC",1350,100,"LTE BAND 3",1,18,-90,-13,-55,23,0,-,-
+QCAINFO: "SCC",154570,1,"NR5G BAND 28",152
```

clndwhr, 2025-05-17:
Perfect!