iamromulan/quectel-rgmii-toolkit #166: [Feature Request] UTF-16 to ASCII

1alessandro1, 2025-05-17:
Hello, when checking the SMS section, phone numbers are encoded in UTF-16, in which a phone number looks like this:

`002B003300390033003400300031003700390033003800390032`

This could be converted to standard ASCII to represent the classical phone number E.164 format, where:

```
-   002B → Unicode U+002B → '+'
-   0033 → '3'
-   0039 → '9'
-   0033 → '3'
-   0034 → '4'
-   0030 → '0'
-   0031 → '1'
```

converting the whole string to `+393401793892`

clndwhr, 2025-05-22:
Tracking 
Bug Fix PR created for this.

https://github.com/iamromulan/quectel-rgmii-toolkit/pull/168

iamromulan, 2025-06-01:
Merged to dev and sdxlemur main
Reopen if not resolved

closed 2025-06-01 by iamromulan

clndwhr, 2025-06-01:
Looks good on my device

[image]