iamromulan/quectel-rgmii-toolkit #175: [SDXPINN] [QuecManager] Authentication screen doesn't restrict access - critical scripts exposed

grapexy, 2025-07-25:
While testing QuecManager, I noticed that the authentication screen doesn’t actually enforce anything. You can access and run critical scripts directly without logging in.

For example, calling this:

```
/cgi-bin/quecmanager/at_cmd/at_queue_client.sh?command=AT%2BQPOWD%3D1&wait=1&timeout=60
```

…will reboot the modem - no auth required.

There are several other .sh scripts exposed in the same way. This could be risky not just if the interface is exposed to the internet, but even on a local network - any rogue device could hit those endpoints and potentially break things or reconfigure the modem silently, or even execute custom shell commands and do full takeover. And since this is running as root, well, opportunities are endless.

Looks like the auth UI is purely cosmetic right now. Definitely worth locking this down before someone runs into trouble.

Thanks!

clndwhr, 2025-07-27:
Code changes are in testing phase at this time.

iamromulan, 2025-07-31:
> So when will be able to install it from Luci/Software ?
> So far i have:
> 
> [image]

It's available as the beta package now on development-SDXPINN, we are still testing it.

If you want to try it sooner edit your OPKG config under custom sources. Edit the URL to be development-SDXPINN instead of SDXPINN and then update lists again.

MiG-41, 2025-07-31:
Thanks , was possible to update to 2.2.4... However something is broken with loging ( rebot not require to login again ) , and was not possible to set bands. So moved back to 2.2.3 again.

clndwhr, 2025-07-31:
@MiG-41, band and cell lock got "broke" on this initial release. The fixes have been applied, were working some processes out for ensuring a good build is created before releasing the fix for this.

MiG-41, 2025-08-01:
Sure, if new fixes aprear for sure i would like to test them.

clndwhr, 2025-08-02:
@MiG-41, 2.2.5 beta is updated.l with these fixes

iamromulan, 2025-08-02:
Will remain open until out of beta

closed 2025-08-31 by clndwhr