iamromulan/quectel-rgmii-toolkit #190: Speedtest on SDXPINN quecmanager-beta

1alessandro1, 2025-08-27:
Hi, I was trying a speedtest from the web ui, but it seems stuck on this screen. Is this a known issue?

[image]

dr-dolomite, 2025-08-27:
Ill check this.

dr-dolomite, 2025-08-27:
For now, can you please try manually running ```speedtest``` using an SSH terminal?

stich86, 2025-08-27:
> For now, can you please try manually running `speedtest` using an SSH terminal to see if its working?

i've found the problem, here in Europe you need to accept also GDPR license, if you add ` --accept-gdpr` into  `/www/cgi-bin/quecmanager/home/speedtest/start_speedtest.sh` will fix the problem :)

dr-dolomite, 2025-08-27:
Added

> > Fixed. Waiting for the patch PR do be merged
> 
> looks like you have missed `--accept-gdpr` as additional flag on the speedtest command :)

closed 2025-08-31 by clndwhr